platformio/platformio-core #189: #includes in included files are not recognized

cqql, 2015-05-02:
Hi,
I wrote a wrapper for `Servo.h`

```cpp
// In motor.h
#include <Servo.>

class Motor {...}
```

```cpp
// In main.cpp
#include "motor.h"
```

When I run `platformio run`, it does not recognize the `#include <Servo.h>` directive and so the `-I` switch for it is missing, when platformio calls avr-g++. I tracked it down to [this line](https://github.com/platformio/platformio/blob/5a162b634182b1cc185978d5381bce7a908a0c72/platformio/builder/tools/platformio.py#L241), but do not know SCons, so I hope, that this is enough to fix it. The problem seems to be, that `env.File` is constructing an `SCons.Node.FS.File` object for the path `"src/motor.h"`, but the `File` constructor manipulates this path to `"scripts/src/motor.h"`, which does not exists. Then `node.get_text_contents()` returns `""`, which obviously does not include the `#include` directive.

ivankravets, 2015-05-03:
Is this a typo in this line ``#include <Servo.>``? Try to replace it to ``#include <Servo.h>``

cqql, 2015-05-03:
No, sorry, that was just a typo on here. I created a [minimal environment](https://github.com/cqql/platformio-bug) to provoke a bug.

Running `platformio run` there produces the following output

```sh
(.env)❯ platformio run
[Sun May  3 23:26:27 2015] Processing uno (targets: upload, platform: atmelavr, board: uno, framework: arduino)
-----------------------------------------------------------------------------
avr-g++ -o .pioenvs/uno/src/Motor.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10603 -DPLATFORMIO=010400 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant .pioenvs/uno/src/Motor.cpp
In file included from .pioenvs/uno/src/Motor.cpp:1:0:
.pioenvs/uno/src/Motor.h:1:19: fatal error: Servo.h: No such file or directory
#include <Servo.h>
^
compilation terminated.
scons: *** [.pioenvs/uno/src/Motor.o] Error 1

```

ivankravets, 2015-05-03:
@valeros Please re-test it. Thanks in advance!

ivankravets, 2015-05-04:
@cqql Thanks for report. I've just reproduced it and will fix tomorrow.

ivankravets, 2015-05-05:
@cqql Please re-test it using [PlatformIO Development Version](http://docs.platformio.org/en/latest/installation.html#development-version).

Thanks in advance!

closed 2015-05-05 by ivankravets

cqql, 2015-05-06:
Now it works as expected.

Thanks a lot :)

commit 440a2b9b (2015-05-05): Fix parsing of includes for PlatformIO Library Dependency Finder // Resolve #189