platformio/platformio-core #190: subdirectories in /src - linker errors

laurentS, 2015-05-03:
My project (on arduino) is structured in modules like:
```
/src/main.ino
/src/file1.[h|cpp]
/src/shared/debug.[h|cpp]
...
```
in file1.h, I have `#include "shared/debug.h"` which worked until platformio auto-upgraded to 1.4.0 (not sure from which version it upgraded, I haven't run it in about a month), and I'm now getting a linker error when compiling:
```
/src/file1.cpp:95: undefined reference to `debug_func(char const*, ...)'
[...]
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/autogen_uno/firmware.elf] Error 1
```
(`debug_func` is in `shared/debug.h`)

I tried tweaking `build_flags` in `platformio.ini` but unless I use absolute paths in the -L and -I options, a "script" prefix is added, so either it doesn't work, or it's not portable.
What happened during the upgrade that broke my project? I don't want to flatten my file structure, is there a proper way to fix platformio.ini for this?

ivankravets, 2015-05-05:
Could I ask you to provide PlatformIO-based project where I can reproduce this issue? 

Thanks in advance!

laurentS, 2015-05-08:
Ok, after testing on simplified code, the problem seems to be that
`/src/shared` is in fact a symbolic link to `../other_directory` (I'm on linux).
if I move `other_directory` to `/src/other_directory`, there is no issue, even with the symbolic link.
The reason for the link is that I was sharing some code between two platformio projects.
I can live with it, so feel free to close the issue, but I wonder what changed in platformio that broke it.

ivankravets, 2015-05-08:
> I can live with it, so feel free to close the issue, but I wonder what changed in platformio that broke it.

Could you provide real example and pack it to archive? I will look on it. Thanks,

laurentS, 2015-05-08:
yes, I just tried to upload here, but github doesn't like it, so I emailed you a tarball instead.
Hope it helps!

ivankravets, 2015-05-08:
Thanks a lot! I've just fixed it. Please install the latest development version http://docs.platformio.org/en/latest/installation.html#development-version

I will release new 1.5.0 in the next week.

closed 2015-05-08 by ivankravets

laurentS, 2015-05-08:
Thanks Ivan! Brilliant work!

commit e960302b (2015-05-08): Fix handling symbolic links within source code directory // Resolve #190