platformio/platformio-core #194: Feature:  run --compile-only

TheNotary, 2015-05-07:
Is there an option to just see if the code compiles without flashing any chips?  This would be helpful for people who are testing what code does and doesn't get past the compile phase.  

I hope I haven't overlooked an option to do this.

valeros, 2015-05-08:
Hi @TheNotary!
`platformio run` builds your code.
`platformio run -t upload` flashes board with your firmware.

ivankravets, 2015-05-08:
Looks like you have created a project with enabled auto-uploading. Just disable ``target = upload`` in your `platformio.ini`.

TheNotary, 2015-05-08:
Ah, thanks, I knew I had to have been missing something.  I commented out the `target = upload` line in my platformio.ini file and can now commence further trial and error at learning type casting =)

closed 2015-05-08 by TheNotary