platformio/platformio-core #195: How to include/use Arduino's standard libraries?

ncolomer, 2015-05-12:
Hello,

Referring to a previous comment [here](https://github.com/platformio/platformio-libmirror/issues/3#issuecomment-74431258), I still face a problem when I want to include an Arduino standard library, in my case, Arduino's `EEPROM.h` library.

I figured out that the library files are present in the platformio path `~/.platformio/packages/framework-arduinoavr/libraries/EEPROM/EEPROM.h`. I imagine they've been downloaded while installing the `atmelavr` platform.

How to include and use those standard libs correctly? 
I expected something simple as with the Arduino IDE where we only need to add `#include <EEPROM.h>`

I'm using the latest platformio stable version 1.4.0 with Eclipse CDT IDE. 
My `platformio.ini` file looks the following:
```
[env:prototype]
platform = atmelavr
framework = arduino
board = nanoatmega328
build_flags = -DDEBUG -Llib
#upload_port = /dev/cu.wchusbserial1410
#targets = upload
```

valeros, 2015-05-12:
Hi @ncolomer!
Simple include like `#include <EEPROM.h>` should be enough.
Maybe you can provide build log or a simple project?

ncolomer, 2015-05-12:
Hi @valeros 
Actually, it seems that the include only runs when it is written in the `main.cpp` file.
When I do the include in another (header) file, the following error appear:
```
$ platformio run
[Tue May 12 21:33:48 2015] Processing dev (platform: atmelavr, build_flags: -DDEBUG, board: nanoatmega328, framework: arduino)
---
avr-g++ -o .pioenvs/dev/src/DeviceID.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_NANO -DDEBUG -DARDUINO=10603 -DPLATFORMIO=010400 -I.pioenvs/dev/SPI -I.pioenvs/dev/RF24 -I.pioenvs/dev/FrameworkArduino -I.pioenvs/dev/FrameworkArduinoVariant .pioenvs/dev/src/DeviceID.cpp
In file included from .pioenvs/dev/src/DeviceID.cpp:2:0:
.pioenvs/dev/src/DeviceID.h:6:20: fatal error: EEPROM.h: No such file or directory
#include <EEPROM.h>
^
compilation terminated.
scons: *** [.pioenvs/dev/src/DeviceID.o] Error 1
=== [ ERROR ] Took 0.53 seconds ===
```
Thus, a workaround should be to do all the includes in the main file (to explicitly tell platformio to add the libraries in the build I imagine), and repeat them in my headers/implem files, WDYT?

ivankravets, 2015-05-12:
PlatformIO scans for includes all source files "*.c, *.cpp, *.h".  It should work for you. 

Could I ask you to share a simple project where I can reproduce this issue?

Thanks in advance!

With best regards, Ivan Kravets
- "Creative comes from talent, and never from knowledge" (c)
_________________________
http://www.ikravets.com

ncolomer, 2015-05-13:
Hi @ivankravets :)
Sure, you can download the zipfile of the project directly from [here](https://dl.dropboxusercontent.com/u/22884360/WeatherMote.zip). Then, simply (un)comment the `#include <EEPROM.h>` line in the `main.cpp` file and run `platformio run` to see the problem.
Cheers,
Nicolas

ivankravets, 2015-05-13:
Thanks! This bug is fixed in the latest ``develop`` branch. You can try it via http://docs.platformio.org/en/latest/installation.html#development-version

P.S; I'm going to release 1.5.0 this week, where this issue will be fixed.

ivankravets, 2015-05-13:
This is a duplicate of https://github.com/platformio/platformio/issues/189

closed 2015-05-13 by ivankravets

ncolomer, 2015-05-13:
Great! Thanks a lot @ivankravets @valeros :)

Jeppedy, 2016-03-03:
Is it broken again in 1.6.5?  I have the exact same issue, with the exact same fix...  If I include the #include in my main ino file, all works fine.  I don't reference anything from that header, but I have to include it.
I only actually reference that header in a library (some cpp file elsewhere) that is USED in my sketch.
Sounds like the exact same problem...

Thoughts?

ivankravets, 2016-03-03:
@Jeppedy please provide the problem project to reproduce this issue

evanlooij, 2016-03-17:
Same problem here, using embedXcode and Xcode to compile a sketch agains Arduino IDE 1.6.7. 
`#include <EEPROM.h>`  leads to

>  error: eeprom.h: No such file or directory

`#include <avr/eeprom.h>`
does allow one to use the AVR functions, i.e. eprom_read_byte (docs at [avr-libc](http://www.nongnu.org/avr-libc/user-manual/group__avr__eeprom.html#ga0ebd0e867b6f4a03d053801d3508f8de))

ivankravets, 2016-03-17:
@evanloooj please `platformio` tool for compilation

tdehaeze, 2016-07-04:
Same issue, I use the platformio CLI to compile and flash.

When I write `#include <EEPROM.h>` in any `.cpp` file expect `main.cpp` it throws an error when I run `platformio run` : `fatal error: EEPROM.h: No such file or directory`.

Same for my own libraries, I have to include them all in my `main.cpp` file in order to work.

tdehaeze, 2016-07-04:
After some research, it's finnally working using [this fix](https://github.com/platformio/platformio/issues/672) (enable deep search for dependent libraries)
Thanks

mintisan, 2016-12-24:
@tdehaeze I also encounter this issue. Can you show me how to fix it with a `platformio.ini` example? 

Thank you.

bitflogger, 2017-01-12:
IDE 1.71, Core 3.2.1, on Windows 7, I have a similar issue, i managed to replicate it. My issue was that I created a file in my project called 'eeprom.h', which I included like this: '#include "eeprom.h"'.
Once I renamed the eeprom.h I saw no issues anymore.

Having said this, I would like to stipulate that this issue is not seen on my linux, command-line box!

ivankravets, 2017-01-12:
@tdehaeze , @mintisan , @bitflogger could you provide a project example to reproduce this issue?

noisymime, 2017-01-18:
I'm seeing exactly the same issue with including EEPROM.h
Have tried with and without the deep search option and it appeared to make no difference (Assuming I've enabled it correctly). 

Copy of the project is attached. This project compiles fine in the Arduino IDE. 
[speeduino copy.zip](https://github.com/platformio/platformio-core/files/715245/speeduino.copy.zip)

ivankravets, 2017-01-18:
@noisymime please add `lib_deps = EEPROM` to `[env:teensy35]`.

noisymime, 2017-01-18:
@ivankravets Thanks Ivan, that fixed it (Needed to be added to `[env:megaatmega2560]` as well)