platformio/platformio-core #196: no prompts please

pelikhan, 2015-05-13:
Please remove all blocking prompts from the command line. It breaks automation scenario easily.

I know there are various settings to disable but prompts are just a bad idea for a command line tool environment. If i wanted prompts, i'd be using an IDE.

ivankravets, 2015-05-13:
How about this issue https://github.com/platformio/platformio/issues/152 ?

Do you recommend to disable prompts by default?

pelikhan, 2015-05-13:
It should be the default. Prompts  might not occur during development time but happen later on the production server due to an update.

ivankravets, 2015-05-14:
I'm not sure that we should link "prompts" with "IDE". We have a lot of console utilities where prompts are used, like ``rm`` with ``--force`` option. Prompts are good for PlatformIO because the allow:

1. Initialise new project with a few questions as for correct project directory, use/or not use automatic uploading and etc.
2. Allow to paginate via search result form ``platformio lib search`` command. Because I can't show ALL result by user request.
3. It allow to avoid problem with internet connection when PlatformIO detect that platform is not installed and ask user to install it. In this case a few hundred megabytes will be used for downloading depending packages.

--------
I implemented ``--force`` option for ALL platformio command which will allow to avoid blocking. See https://github.com/platformio/platformio/issues/152

Like, ``platformio --force COMMAND OPTIONS``

pelikhan, 2015-05-14:
Ok will add force. I find prompts in all command lines utterly annoying, I guess that's just me :)

ivankravets, 2015-05-14:
> I guess that's just me

I hope you will like PlatformIO 2.0 :smile: Thanks a lot for your ideas and interest.

closed 2015-05-14 by ivankravets