platformio/platformio-core #202: Integration with Blynk library

ccaroon, 2015-05-15:
When trying to compile one of the Blynk (blynk.cc) simple examples I get the following error. The odd thing is that I can't figure out why `platformio` is trying to use the EngduinoSD library in the first place

```C++
#include <SPI.h>
#include <Adafruit_CC3000.h>
#include <BlynkSimpleTinyDuino.h>

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "YourAuthToken";

void setup()
{
  //Serial.begin(9600);
  Blynk.begin(auth, "ssid", "pass", WLAN_SEC_WPA2);
}

void loop()
{
  Blynk.run();
}
```

Compile Error:

avr-g++ -o .pioenvs/autogen_tinyduino/EngduinoSD/EngduinoSD.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=8000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_LILYPAD -DARDUINO=10603 -I.pioenvs/autogen_tinyduino/Adafruit_CC3000_ID17/utility -I.pioenvs/autogen_tinyduino/Blynk/utility -I.pioenvs/autogen_tinyduino/Ethernet/utility -I.pioenvs/autogen_tinyduino/WiFi/utility -I.pioenvs/autogen_tinyduino/EngduinoSD/utility -I.pioenvs/autogen_tinyduino/SPI -I.pioenvs/autogen_tinyduino/Adafruit_CC3000_ID17 -I.pioenvs/autogen_tinyduino/Blynk -I.pioenvs/autogen_tinyduino/Ethernet -I.pioenvs/autogen_tinyduino/WiFi -I.pioenvs/autogen_tinyduino/Bridge -I.pioenvs/autogen_tinyduino/EngduinoSD -I.pioenvs/autogen_tinyduino/FrameworkArduino -I.pioenvs/autogen_tinyduino/FrameworkArduinoVariant .pioenvs/autogen_tinyduino/EngduinoSD/EngduinoSD.cpp
In file included from .pioenvs/autogen_tinyduino/EngduinoSD/EngduinoSD.cpp:20:0:
.pioenvs/autogen_tinyduino/EngduinoSD/EngduinoSD.h:19:22: fatal error: Engduino.h: No such file or directory
#include <Engduino.h>
^
compilation terminated.

ivankravets, 2015-05-15:
PlatformIO Library Finder lookups for libs in alphabetical order. "EngduinoSD" contains SPI.h file, which was detected automatically. In your case you need to ignore EngduinoSD library, then PlatformIO Library Finder will use core <SPI> lib. 

Please look into http://docs.platformio.org/en/latest/projectconf.html#ignore-libs

Did it help you?

With best regards, Ivan Kravets
- "Creative comes from talent, and never from knowledge" (c)
_________________________
http://www.ikravets.com

ccaroon, 2015-05-15:
Yes, so adding the `ignore_libs` works to get me past the issue. Thanks, I had not seen that setting.
However, I don't understand your comment about SPI.h being part of EngduinoSD.

SPI looks to be a separate library. Here my list of libs. You'll notice SPI toward the end of the second line.

$HOME/.platformio/packages/framework-arduinoavr/libraries:
```
Audio/			EngduinoLight/		Ethernet/		SD/			TFT/
Bridge/			EngduinoMagnetometer/	Firmata/		SPI/			Temboo/
EEPROM/			EngduinoProtocol/	GSM/			Scheduler/		USBHost/
EngduinoAccelerometer/	EngduinoSD/		LiquidCrystal/		Servo/			WiFi/
EngduinoButton/		EngduinoTD/		RobotIRremote/		SoftwareSerial/		Wire/
EngduinoIR/		EngduinoThermistor/	Robot_Control/		SpacebrewYun/		__cores__/
EngduinoLEDs/		Esplora/		Robot_Motor/		Stepper/
```

ivankravets, 2015-05-15:
Give me please a list of files from this directory ``$HOME/.platformio/packages/framework-arduinoavr/libraries/EngduinoSD``

ccaroon, 2015-05-16:
[~/.platformio/packages/framework-arduinoavr/libraries]>ls -R EngduinoSD/
EngduinoSD/:
keywords.txt*  library.properties*  src/

EngduinoSD/src:
EngduinoSD.cpp*  EngduinoSD.h*  File.cpp*  README.txt*  SD.cpp*  SD.h*
utility/

EngduinoSD/src/utility:
FatStructs.h*  Sd2Card.cpp*  Sd2Card.h*  Sd2PinMap.h*  SdFat.h*
SdFatmainpage.h*  SdFatUtil.h*  SdFile.cpp*  SdInfo.h*  SdVolume.cpp*

ivankravets, 2015-05-18:
PlatformIO tries to build library which has been detected by PlatformIO Library Dependency Finder (PLDF). You use [Blynk](http://platformio.org/#!/lib/show/415/Blynk) library which requires ``SD.h`` in https://github.com/blynkkk/blynk-library/blob/master/WidgetSD.h

Nevertheless, PLDF analyses libraries in alphabetical order and ``EngduinoSD`` was automatically picked up (no ``SD``). Please look into ``~/.platformio/packages/framework-arduinoavr/libraries``

So, how about to rename all Engduino libraries with prefix ``ZZ_Engduino*`` ?

ivankravets, 2015-05-18:
@valeros Please rename all "Engduino" libraries to ``ZZ_Engduino*`` and re-pack Arduino framework. Also, please report here when it will be done and ask @ccaroon to make ``platformio update`` and re-test it.

Thanks in advance!

valeros, 2015-05-18:
I updated framework. @ccaroon Please, try new version.

ccaroon, 2015-05-19:
I updated and the problem is fixed.

However, it's still not clear to me why the Engduino code is being picked up and trying to be built. The Blynk code, as far as I can tell, does not depend on or use Engduino code.

Thanks again for your help and attention to this matter.

ivankravets, 2015-05-19:
@ccaroon please read my comment above. As I mentioned, ``Blynk`` library uses ``SD.h`` in https://github.com/blynkkk/blynk-library/blob/master/WidgetSD.h

``SD`` is native library from Arduino Framework. However, ``EngduinoSD`` had more higher priority (letter E < S, order). Now, ``EngduinoSD`` has been renamed to ``ZZ_EngduinoSD`` and has more lower priority before all native libraries.

closed 2015-05-21 by ivankravets