platformio/platformio-core #203: "undefined reference" issue

nicc777, 2015-05-16:
Hi - I'm very rusty with the coding side, so please take that into consideration :-)

I am testing platformio with the examples from the SparkFun Inventors Kit. I am stuck on example 6 (photo resistor example).

I get the following error when I do `$ platformio run`:

    .pioenvs/autogen_sparkfun_redboard/src/circuit6.o: In function `manualTune':
    /home/user/PlatformIO_Projects/Sparkfun_Redboard_Circuit06/.pioenvs/autogen_sparkfun_redboard/src/circuit6.c:25: undefined reference to `map'

src/circuit6:

        #include "Arduino.h"

        const int sensorPin = 0;
        const int ledPin = 9;

        int lightLevel, high = 0, low = 1023;
  
        void setup()
        {
          pinMode(ledPin, OUTPUT);
        }

        void loop()
        {
          lightLevel = analogRead(sensorPin);
          // lightLevel = map(lightLevel, 0, 1023, 0, 255);
          // lightLevel = constrain(lightLevel, 0, 255);
          manualTune();  // manually change the range from light to dark
          //autoTune();  // have the Arduino do the work for us!
          analogWrite(ledPin, lightLevel);
        }

        void manualTune()
        {
          lightLevel = map(lightLevel, 0, 1023, 0, 255);
          lightLevel = constrain(lightLevel, 0, 255);
        }

        void autoTune()
        {
          if (lightLevel < low)
          {
            low = lightLevel;
          }
          if (lightLevel > high)
          {
            high = lightLevel;
          }
          lightLevel = map(lightLevel, low+30, high-30, 0, 255);
          lightLevel = constrain(lightLevel, 0, 255);
        }

And `platform.ini`:

        [env:autogen_sparkfun_redboard]
        platform = atmelavr
        framework = arduino
        board = sparkfun_redboard
        # targets = upload
        #use_libs = Arduino
        build_flags = -Q

If it helps, I did see the following in the output and from my limited understanding, this should have worked:

        Configure with --enable-checking=release to enable checks.
        avr-g++ -o .pioenvs/autogen_sparkfun_redboard/FrameworkArduino/WMath.o -c -fno-exceptions -fno-threadsafe-statics -Q -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10603 -I.pioenvs/autogen_sparkfun_redboard/FrameworkArduino -I.pioenvs/autogen_sparkfun_redboard/FrameworkArduinoVariant .pioenvs/autogen_sparkfun_redboard/FrameworkArduino/WMath.cpp                                                                                             char* itoa(int, char*, int) char* ltoa(long int, char*, int) char* utoa(unsigned int, char*, int) char* ultoa(long unsigned int, char*, int) void randomSeed(unsigned int) long int random(long int) long int random(long int, long int) long int map(long int, long int, long int, long int, long int) unsigned int makeWord(unsigned int) unsigned int makeWord(unsigned char, unsigned char)              
        Analyzing compilation unit
        Performing interprocedural optimizations
        <*free_lang_data> <visibility> <early_local_cleanups> <*free_inline_summary> <emutls> <whole-program> <profile_estimate> <cp> <inline> <pure-const> <static-var>Assembling func tions:
        void randomSeed(unsigned int) long int random(long int) long int random(long int, long int) long int map(long int, long int, long int, long int, long int) unsigned int makeWord(unsigned int) unsigned int makeWord(unsigned char, unsigned char)

Am I doing something obvious wrong? It works perfectly when I use the Arduino IDE

ivankravets, 2015-05-18:
``long map(long x, long in_min, long in_max, long out_min, long out_max)`` function is located in ``WMath.cpp`` of Arduino Core. However, you are trying to use this function from ``*.c`` file. Please re-name your file to ``*.cpp`` or to ``.ino``. Like:

```c
const int sensorPin = 0;
const int ledPin = 9;

int lightLevel, high = 0, low = 1023;

void setup()
{
  pinMode(ledPin, OUTPUT);
}

void loop()
{
  lightLevel = analogRead(sensorPin);
  // lightLevel = map(lightLevel, 0, 1023, 0, 255);
  // lightLevel = constrain(lightLevel, 0, 255);
  manualTune();  // manually change the range from light to dark
  //autoTune();  // have the Arduino do the work for us!
  analogWrite(ledPin, lightLevel);
}

void manualTune()
{
  lightLevel = map(lightLevel, 0, 1023, 0, 255);
  lightLevel = constrain(lightLevel, 0, 255);
}

void autoTune()
{
  if (lightLevel < low)
  {
    low = lightLevel;
  }
  if (lightLevel > high)
  {
    high = lightLevel;
  }
  lightLevel = map(lightLevel, low+30, high-30, 0, 255);
  lightLevel = constrain(lightLevel, 0, 255);
}
```

closed 2015-05-18 by ivankravets

nicc777, 2015-05-18:
Works perfectly now! Thanks!