platformio/platformio-core #204: Add Armstrap variations as available boards

beriberikix, 2015-05-17:
I'm not sure if there is an automated process to upload boards to the central repository, but it would be great to have the @armstrap on there (as specified in https://github.com/platformio/platformio/issues/175#issuecomment-94838565.) Note that there are [3 variations](http://docs.armstrap.org/en/latest/hardware-overview.html#armstrap-eagle) of with differently-sized on-chip memory.

ivankravets, 2015-05-18:
You can try new boards with PlatformIO 2.0a1 using http://docs.platformio.org/en/latest/installation.html#development-version

closed 2015-05-18 by ivankravets

beriberikix, 2015-05-23:
I got this working with the stable 2.0. However, I still needed to include an upload.gdb and add the gdb protocol to the `.ini.` Is that as expected?

https://github.com/beriberikix/armstrap-gcc-barebones/commit/9648df3e1ac9bbe45b7d1ba14c2306c3f23ba03d

ivankravets, 2015-05-24:
> beriberikix/armstrap-gcc-barebones@9648df3

Yes, it's normal behavior of PlatformIO. This approach allows you to make own GDB commands list. 

With best regards, Ivan Kravets
- "Creative comes from talent, and never from knowledge" (c)
_________________________
http://www.ikravets.com

commit 5d547271 (2015-05-18): Update docs with the new boards // Resolve #204