platformio/platformio-core #208: Inconsistency with src_dir

gandy92, 2015-05-20:
My `platformio.ini` declares two environments dealing with seperate src directories:
```
[env:asksin]
platform = atmelavr
framework = arduino
board = pro8MHzatmega328
src_dir = src-asksin
[env:swap]
platform = atmelavr
framework = arduino
board = panStampAVR
src_dir = src-swap
```
Now, if I try to compile with `platformio run -e swap` the output starts with
```
[Wed May 20 16:15:29 2015] Processing swap (platform: atmelavr, board: panStampAVR, src_dir: src-swap, framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
```
(note the "`src_dir: src-swap`"). 

Compilation however fails, because none of the files in `src-swap` are copied to the `.pioenvs/swap` directory. 

Manually copying the contents from `src-swap/` to `src/` helps compiling, but this defeats the purpose of declaring an alternate `src_dir`.

ivankravets, 2015-05-20:
`src_dir` is available for `[platformio]` section, but not for `[env:***]`. 

If you have different source code per environment then you have different projects. The main idea of build environments is to build the same code with the different platforms/boards

gandy92, 2015-05-20:
Agreed. In that case, I have a different problem: 

The two environments are used to build the same project for a panstamp board with support for different RF protocols:
* one is provided by the swap library shipped with the panstamp-avr framework
* the other is provided by lib 179 (NewAskSin) and requires to build without the panstamp-avr framework since NewAskSin provides files that have identical names to some from the panstamp-avr framework

In both cases, the support for the specific protocol is a minor fraction of the project, so most of the project's code would be duplicated if I make two projects.

The first solution I tried was to include all protocol specific code in `#ifdef` statements to make compilation depend on the active environment (testing for ARDUINO_AVR_PRO or PANSTAMP_AVR). This however did not work because platformio does not honor these statements when checking for include files. Thus, when compiling the swap evironment, the NewAskSin library is still being compiled, even though it is not required. Since this library does not compile with the panstamp-avr framework, compilation fails.

Next approach I tried was moving all protocol specific code to different src-directories and the rest of the code to a project-specific lib. This also fails, since obviously, the src_dir statement is global to the project.

Next thing I can try is to move the protocol specific stuff to two different project-specific libraries and test if environment specific includes work this way.

Any ideas how to solve this while keeping all code in one project is highly appreciated.

ivankravets, 2015-05-20:
How about first solution with `#ifdef` and `ignore_libs` per environment?

gandy92, 2015-05-21:
Wow, thanks! I completely missed that, but of course, this works elegantly well :smile:

I'll have to read the changelog of the last two releases to see what else I missed, the last few weeks were just too busy...

closed 2015-05-21 by gandy92

ivankravets, 2015-05-22:
> I'll have to read the changelog of the last two releases to see what else I missed, the last few weeks were just too busy...

We are working on PlatformIO 2.0. It's ready on 99%. I hope that I release it in the nearest days :)