platformio/platformio-core #211: Using C++ with the STM32F4 Discovery causes linker errors, what am I doing wrong?

apvanzanten, 2015-05-23:
I've created a project using the **platformio init --board=disco_f407vg** command, and then proceeded to write a main.cpp. Everything seemed to work fine at first but when I try to use **new** or **delete** I get linker errors. Specifically I have a line **int * testInt = new int;** which then causes **main.cpp:12: undefined reference to `operator new(unsigned int)'** (where line 12 is the line containing the new call).

I'm guessing I still have to add some setting to the platform.ini to configure it for full C++ support, but I can't find anything about such a setting in the documentations. What am I missing?

Thanks for the help (also thanks for building this software, I don't think I've ever had a toolchain and bsp install this quickly and easily!)

valeros, 2015-05-25:
Hello @apvanzanten.

I have updated ldscript for stm32f4, so now you can use PlatformIO develop branch or stable 2.0.0 with this `platformio.ini:`
```
[env:disco_f407vg]
platform = ststm32
framework = cmsis
board = disco_f407vg
build_flags = -lstdc++ -lnosys
```

closed 2015-05-25 by valeros

apvanzanten, 2015-05-25:
Sweet, thanks for the quick fix! :+1:

commit 97de38d6 (2015-05-25): Add c++ support for ststm32 platform // Resolve #211
commit b6a0a0f7 (2015-05-25): Merge pull request #213 from valeros/develop

Add c++ support for ststm32 platform // Resolve #211