platformio/platformio-core #214: teensy arduino linking problem after upgrade to PIO2.0

gandy92, 2015-05-25:
Since upgrading to 2.0, I get an error when compiling a project for teensy3.1-arduino:

'''
../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: cannot open linker script file /home/andy/.platformio/packages/ldscripts/mk20dx256.ld: No such file or directory
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/teensy31/firmware.elf] Error 1
'''

already deleted .pioenv and even reinstalled everything from scratch, but to no avail.

what can I do?

valeros, 2015-05-25:
Hi @gandy92

It seems ldscript package has been updated incorrectly.
This file `mk20dx256.ld` should be in this directory: `/home/user/.platformio/packages/ldscripts` 
If it's not there, try to update your packages via `platformio update`.
If this does not help, delete your `appstate.json` file from `/home/user/.platformio` and install teensy platform again.

gandy92, 2015-05-25:
Thanks, that helped - apparently, a few file permissions in ~/.platformio/packages got corrupted during a previous `sudo platformio upgrade`. After fixing the file permissions, deleting the appstate.json file and installing teensy platform during the next `platformio run`, my build system is back to normal.

ivankravets, 2015-05-25:
@gandy92 Im glad to hear you! :blush: 

Since PlatformIO 2.0 each upgrade operation to the new version will force ``platformio platforms update`` automatically. It should allow to avoid the similar problems like yours.

gandy92, 2015-05-25:
I'm not sure but I gather the problem was a `sudo platform upgrade` which not only upgraded platformio in /usr/local/ with root permissions, but also updated libs and packages residing in $HOME/.platformio with root permissions. This left some of the files (including packages/ldscripts) owned by root instead of myself, making it impossible to update the files or even whole subdirectories without `sudo`.

Next time I run into this kind of problem I know I have to check permissions in my `~/.platformio`. 

Would it be possible to check if platformio update is running with root permissions and fix file ownership in non-system directories?

ivankravets, 2015-05-25:
> Would it be possible to check if platformio update is running with root permissions and fix file ownership in non-system directories?

I've just fixed it. PlatformIO will not run updating process for the packages/libs when user runs ``platformio upgrade`` command. 

P.S: Since PlatformIO 2.0 automatic library/package updates were disabled by default.

closed 2015-05-25 by ivankravets

gandy92, 2015-05-25:
Perfect, thanks! :+1: 

Must have happened some while ago. I remember having encountered problems during some updates but each time, rerunning `platformio update` appeared to resolve the problem. Now, in retrospect, what really might have happened was that the appstate.json was updated without the complete update ever happening.

ivankravets, 2015-05-25:
@gandy92 what do you thing about new feature [Continuous Integration](http://docs.platformio.org/en/latest/ci/index.html) in PlatformIO 2.0? 

See examples:
* https://travis-ci.org/felis/USB_Host_Shield_2.0
* https://travis-ci.org/ivankravets/Arduino-IRremote

commit 76497694 (2015-05-25): Disable packages/libs updating while "upgrade" operation // Resolve #214