platformio/platformio-core #216: Arduino lib no more resolved after update to v2

ncolomer, 2015-05-25:
Hi there!

I just updated platformio to the brand new v2 (thanks for this release!) and tried the cool `init --ide eclipse` feature.
But it appears that libs such as `avr/io.h` or `stdlib.h` (included in `Arduino.h`) are no more resovled by eclipse (even after clean / rebuild), resulting in build error in the IDE whereas CLI build is running well.

See following picture:
[image]

I [zipped](https://dl.dropboxusercontent.com/u/22884360/WeatherMote.zip) the project if you want to test by yourself. Maybe I've done things the wrong way, please tell me.

Cheers,
Nicolas

ivankravets, 2015-05-25:
I know it, thanks. Please add ``__AVR_ATmega328__`` symbol. 
[image]

More details here http://www.ikravets.com/wp-content/uploads/2014/06/eclipse_setup_project_symbols.png

P.S: I'll fix it tomorrow.

ncolomer, 2015-05-26:
As usual, thanks for your quick answer @ivankravets :)

It seems this is not enough to fix the issue, even after clean/rebuild the project from ide and from CLI, plus restarted eclipse. 

[image]

I also re-checked your blog post and it seems a directory was also missing: `${HOME}/.platformio/packages/toolchain-atmelavr/avr/include`. So I added it to all languages and, and it fixed the issue!

ivankravets, 2015-05-26:
> As usual, thanks for your quick answer @ivankravets :)

Thank you too for the report and retweets!!! :+1: 

> I also re-checked your blog post and it seems a directory was also missing: ${HOME}/.platformio/packages/toolchain-atmelavr/avr/include. So I added it to all languages and, and it fixed the issue!

This issue was fixed in #210 

> Please add __AVR_ATmega328__ symbol. 

This symbol is required for ``Serial`` and etc lib. Basing on MCU framework detects how much UARTs device has. I'll fix it in 2.0.1 release.

commit 1f373b1e (2015-05-26): Handle AVR Symbols when initialising project for IDE // Resolve #216
closed 2015-05-26 by ivankravets