platformio/platformio-core #219: File nano.specs missing when compiling for DUE/atmelsam on BeagleBone Black

eliasse, 2015-05-27:
Hello,

First, thank you for an incredible tool!

I am trying to compile code for an Arduino DUE on a BeagleBone Black. The code compiles fine using PlatformIO on my Linux Mint machine. 

___________________________________________
arm-none-eabi-g++ -o .pioenvs/autogen_due/firmware.elf -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m3 --specs=nano.specs -Wl,--entry=Reset_Handler -Wl,--start-group -T /home/debian/.platformio/packages/ldscripts/sam3x8e.ld .pioenvs/autogen_due/src/TCM_CLASS.o .pioenvs/autogen_due/src/tmp_ino_to.o -L/home/debian/.platformio/packages/framework-arduinosam/variants/arduino_due_x -L.pioenvs/autogen_due -Wl,--start-group -lc -lgcc -lm .pioenvs/autogen_due/libFrameworkArduinoVariant.a .pioenvs/autogen_due/libFrameworkArduino.a -lsam_sam3x8e_gcc_rel -Wl,--end-group
arm-none-eabi-g++: **error: nano.specs: No such file or directory**
scons: *** [.pioenvs/autogen_due/firmware.elf] Error 1
___________________________________________

I get the exact same kind of error if i try to compile it on a Raspberry Pi (Arm v6).
Naively, I tried copying the **nano.specs** files from the ~/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/lib directory on my Mint machine to the corresponding folder on the BeagleBone. This clears the "no-such-file" error but gives a linker error instead. 

Very grateful for any help on this matter!

Best Regards

ivankravets, 2015-05-29:
@eliasse please install [development version of PlatformIO](http://docs.platformio.org/en/latest/installation.html#development-version) and update existing packages via ``platformio platforms update``.

Did it help you?

closed 2015-05-29 by ivankravets

eliasse, 2015-05-29:
Yes, it works!

Thanks for your help.
I still have a linker problem but I might create a separate issue for that if I can't figure it out myself.

ivankravets, 2015-05-29:
If your linker problem isn't linked with the current issue and "nano.specs", please create separate issue.

Thanks in advance!

commit 1debe847 (2015-05-29): Disabled "nano.specs" for ARM-based platforms // Resolve #219