platformio/platformio-core #220: C/C++ std libs no more resolved by Eclipse IDE

ncolomer, 2015-05-27:
The standard C/C++ libraries include are no more resolved by Eclipse CDT IDE after initializing the project with `platformio init --ide eclipse` (eg. `stdarg.h`, `stdbool.h` or `stddef.h`). 

It seems to be a regression since I did not face the problem when settings the project configuration manually according to @ivankravets [blog post], cf image below:

[image]

ivankravets, 2015-05-27:
Thanks again for the report :+1: . Please try the latest build http://docs.platformio.org/en/latest/installation.html#development-version

Did it help you?

closed 2015-05-27 by ivankravets

ncolomer, 2015-05-29:
Hi @ivankravets, a9a9545 fixed the issue, thanks!

commit a9a9545b (2015-05-27): Fix resolving of C/C++ std libs by Eclipse IDE // Resolve #220