platformio/platformio-core #221: Upload to Arduino embedded custom board

ncolomer, 2015-05-29:
I have build a custom board powered with an `atmega328p`. Currently, I program it over ISP by uploading platformio's generated `.hex` file via an USBasp clone and [AVRFuses](http://www.vonnieda.org/AVRFuses/) software.

I'm looking for doing the upload task directly via platformio using the `upload` target. I saw in the [Project Configuration File `platformio.ini`](http://docs.platformio.org/en/latest/projectconf.html#examples) documentation that some parameters (such as `board_mcu`, `board_f_cpu`, `upload_protocol` or `upload_speed`) could help. But it is ever possible? In addition, it is possible to set MCU fuses?

The two following examples mentioned in the page above seem close to what I want to achieve:
> 4 - Platform atmelavr: Embedded board that is based on ATmega168 MCU with “arduino” bootloader
> 5 - Upload firmware via USB programmer (USBasp) to Platform atmelavr microcontrollers

But I can't make them running: either I mention the `framework = arduino` entry in `platformio.ini` file and I have the following error:
```bash
$ platformio run                                                                                                                                                                                     1 ↵
[Fri May 29 14:02:51 2015] Processing WeatherMote_v01 (upload_protocol: usbasp -B5, framework: arduino, build_flags: -Llibs -DDEBUG -Wl,-u,vfprintf -lprintf_flt, board_f_cpu: 16000000L, platform: atmelavr, upload_speed: 19200, board_mcu: atmega328p, targets: upload)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
TypeError: argument of type 'NoneType' is not iterable:
File "/usr/local/lib/python2.7/site-packages/platformio/builder/main.py", line 122:
SConscript(env.subst("$BUILD_SCRIPT"))
File "/usr/local/lib/scons-2.3.0/SCons/Script/SConscript.py", line 609:
return method(*args, **kw)
File "/usr/local/lib/scons-2.3.0/SCons/Script/SConscript.py", line 546:
return _SConscript(self.fs, *files, **subst_kw)
File "/usr/local/lib/scons-2.3.0/SCons/Script/SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "/usr/local/lib/python2.7/site-packages/platformio/builder/scripts/atmelavr.py", line 92:
target_elf = env.BuildFirmware()
File "/usr/local/lib/scons-2.3.0/SCons/Environment.py", line 223:
return self.method(*nargs, **kwargs)
File "/usr/local/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 28:
env.BuildFramework()
File "/usr/local/lib/scons-2.3.0/SCons/Environment.py", line 223:
return self.method(*nargs, **kwargs)
File "/usr/local/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 137:
if framework in env.get("BOARD_OPTIONS", {}).get("frameworks"):
```

or I do not mention it and the compilation fails since I'm relying on Arduino libs in my program:
```bash
 platformio run
[Fri May 29 14:02:36 2015] Processing WeatherMote_v01 (upload_protocol: usbasp -B5, build_flags: -Llibs -DDEBUG -Wl,-u,vfprintf -lprintf_flt, board_f_cpu: 16000000L, platform: atmelavr, upload_speed: 19200, board_mcu: atmega328p, targets: upload)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
avr-g++ -o .pioenvs/WeatherMote_v01/src/main.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DDEBUG -DPLATFORMIO=020003 -I.pioenvs/WeatherMote_v01/RF24 src/main.cpp
src/main.cpp:2:21: fatal error: Arduino.h: No such file or directory
#include <Arduino.h>
^
compilation terminated.
scons: *** [.pioenvs/WeatherMote_v01/src/main.o] Error 1
```

Any hints, idea (or simply can't do this way)?

valeros, 2015-05-29:
Hi @ncolomer !

Your board is similar to Ardino Uno, so just try to add `board = uno` in your `platformio.ini`:

```
[env:WeatherMote_v01]
platform = atmelavr
framework = arduino
board = uno
board_mcu = atmega328p
upload_protocol=usbasp -B5
build_flags = -Llibs -DDEBUG -Wl,-u,vfprintf -lprintf_flt
board_f_cpu = 16000000L
upload_speed = 19200
targets = upload
```

ivankravets, 2015-05-29:
@ncolomer please upgrade PlatformIO to ``2.0.3.dev1`` from development branch and test with this ``platformio.ini``:
```ini
[env:WeatherMote_v01]
platform = atmelavr
framework = arduino
board = uno
upload_protocol=usbasp -B5
build_flags = -Llibs -DDEBUG -Wl,-u,vfprintf -lprintf_flt
targets = upload
```

P.S: Thanks a lot for your reported issues. It allows us to improve PlatformIO.

ncolomer, 2015-05-29:
Thanks @valeros @ivankravets for your answers!

Well I really enjoy using platformio for my electronic projects since i'm also a dev and I know build tools are always welcome, especially when dealing with C/C++ and embedded stuff! PlatformIO simplify a lot of things and gives a clear and featureful framework for hobbyists like me. I really want to use it in all my projects... and give as much help as I can :) You're doing great, guys!

So, now it compiles, but upload fails:

```
avrdude: Version 6.0.1, compiled on Dec 16 2013 at 17:26:24
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2009 Joerg Wunsch

System wide configuration file is "/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude.conf"

... ATmega328P conf stuff ...

Programmer Type : usbasp
Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: set SCK frequency to 187500 Hz
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f
avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as DF
avrdude: safemode: efuse reads as 7
avrdude: reading input file ".pioenvs/WeatherMote_v01/firmware.hex"
avrdude: writing flash (8574 bytes):

Writing | ################################################## | 100% 5.44s

avrdude: 8574 bytes of flash written
avrdude: verifying flash memory against .pioenvs/WeatherMote_v01/firmware.hex:
avrdude: load data flash data from input file .pioenvs/WeatherMote_v01/firmware.hex:
avrdude: input file .pioenvs/WeatherMote_v01/firmware.hex contains 8574 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 4.29s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0040
0xe0 != 0xef
avrdude: verification error; content mismatch

avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as DF
avrdude: safemode: efuse reads as 7
avrdude: safemode: Fuses OK (H:07, E:DF, L:FF)

avrdude done.  Thank you.

scons: *** [upload] Error 1
```

> avrdude: safemode: Fuses OK (H:07, E:DF, L:FF)

Fuses are what I set previously using [AVRFuses](http://www.vonnieda.org/AVRFuses/) software. I suppose they are not overridden. What if I want to specify others? Do I still need stuff like AVRFuses?

> avrdude: set SCK frequency to 187500 Hz

I have a doubt with this frequency (we program Arduino at 19200 bauds, no?), but I suppose it's the default value.

> avrdude: verification error, first mismatch at byte 0x0040

So there is a mismatch during verification that let me think the writing goes wrong.

The following command is run by PlatformIO (and its embedded avrdude): `"/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude" -v -D -p atmega328p -C "/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude.conf" -c usbasp -B5 -U flash:w:.pioenvs/WeatherMote_v01/firmware.hex:i`. Do you know what means the `-B5`?

And the following command is what is run by AVRFuses when burning a `.hex` files (I installed avrdude 6.1 via homebrew): `/usr/local/Cellar/avrdude/6.1/bin/avrdude -c usbasp-clone -p ATmega328P -qq -U flash:w:/Users/ncolomer/Developer/Workspace/electronics/HomeMote/WeatherMote/.pioenvs/WeatherMote_v01/firmware.hex`

There is some difference, including CLI options, but I can't really explain them.

ivankravets, 2015-05-29:
Have you tried this?
```ini
[env:WeatherMote_v01]
platform = atmelavr
framework = arduino
board = uno
upload_protocol=usbasp-clone
build_flags = -Llibs -DDEBUG -Wl,-u,vfprintf -lprintf_flt
targets = upload
```

ncolomer, 2015-05-30:
Nop :( The `usbasp-clone` upload protocol is correctly recognized but the error remains the same:

> avrdude: verification error, first mismatch at byte 0x0040
> 0xe0 != 0xef
> avrdude: verification error; content mismatch

Full log output:
```
avrdude: Version 6.0.1, compiled on Dec 16 2013 at 17:26:24
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2009 Joerg Wunsch

System wide configuration file is "/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude.conf"

Using Port                    : usb
Using Programmer              : usbasp-clone
AVR Part                      : ATmega328P
Chip Erase delay              : 9000 us
PAGEL                         : PD7
BS2                           : PC2
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 200
StabDelay                     : 100
CmdexeDelay                   : 25
SyncLoops                     : 32
ByteDelay                     : 0
PollIndex                     : 3
PollValue                     : 0x53
Memory Detail                 :

Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

Programmer Type : usbasp
Description     : Any usbasp clone with correct VID/PID

avrdude: auto set sck period (because given equals null)
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f
avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as DF
avrdude: safemode: efuse reads as 7
avrdude: reading input file ".pioenvs/WeatherMote_v01/firmware.hex"
avrdude: writing flash (8574 bytes):

Writing | ################################################## | 100% 5.42s

avrdude: 8574 bytes of flash written
avrdude: verifying flash memory against .pioenvs/WeatherMote_v01/firmware.hex:
avrdude: load data flash data from input file .pioenvs/WeatherMote_v01/firmware.hex:
avrdude: input file .pioenvs/WeatherMote_v01/firmware.hex contains 8574 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 4.29s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0040
0xe0 != 0xef
avrdude: verification error; content mismatch

avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as DF
avrdude: safemode: efuse reads as 7
avrdude: safemode: Fuses OK (H:07, E:DF, L:FF)

avrdude done.  Thank you.

scons: *** [upload] Error 1
```

Here is the output when I use AVRFuses:

```
Programming flash...

/usr/local/Cellar/avrdude/6.1/bin/avrdude -c usbasp-clone -p ATmega328P -qq -U flash:w:/Users/ncolomer/Developer/Workspace/electronics/HomeMote/WeatherMote/.pioenvs/WeatherMote_v01/firmware.hex

avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
SUCCESS

Verifying flash...

/usr/local/Cellar/avrdude/6.1/bin/avrdude -c usbasp-clone -p ATmega328P -qq -U flash:v:/Users/ncolomer/Developer/Workspace/electronics/HomeMote/WeatherMote/.pioenvs/WeatherMote_v01/firmware.hex

avrdude: warning: cannot set sck period. please check for usbasp firmware update.
SUCCESS
```

ivankravets, 2015-05-30:
Thanks, let's try a few variants:

a) 
```shell
"/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude" -v -p atmega328p -C "/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude.conf" -c usbasp-clone -U flash:w:.pioenvs/WeatherMote_v01/firmware.hex:i
```
b)
```shell
"/usr/local/Cellar/avrdude/6.1/bin/avrdude" -v -p atmega328p -C "/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude.conf" -c usbasp-clone -U flash:w:.pioenvs/WeatherMote_v01/firmware.hex:i
```
c)
```shell
"/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude" -v -p atmega328p -C "/usr/local/Cellar/avrdude/6.1/etc/avrdude.conf" -c usbasp-clone -U flash:w:.pioenvs/WeatherMote_v01/firmware.hex:i
```
d)
```shell
"/usr/local/Cellar/avrdude/6.1/bin/avrdude" -v -c usbasp-clone -p ATmega328P -U flash:w:.pioenvs/WeatherMote_v01/firmware.hex:i
```

All commands should be performed from project directory where ``platformio.ini`` is located.

ncolomer, 2015-06-01:
(sorry for the delay) Here are the output of commands

a) is ok
```
$ "/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude" -v -p atmega328p -C "/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude.conf" -c usbasp-clone -U flash:w:.pioenvs/WeatherMote_v01/firmware.hex:i

avrdude: Version 6.0.1, compiled on Dec 16 2013 at 17:26:24
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude.conf"
         User configuration file is "/Users/ncolomer/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp-clone
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : Any usbasp clone with correct VID/PID

avrdude: auto set sck period (because given equals null)
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f
avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as DF
avrdude: safemode: efuse reads as 7
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: reading input file ".pioenvs/WeatherMote_v01/firmware.hex"
avrdude: writing flash (10562 bytes):

Writing | ################################################## | 100% 6.52s

avrdude: 10562 bytes of flash written
avrdude: verifying flash memory against .pioenvs/WeatherMote_v01/firmware.hex:
[... 184 lines cut ...]
avrdude: auto set sck period (because given equals null)
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: reading input file ".pioenvs/WeatherMote_v01/firmware.hex"
avrdude: writing flash (10562 bytes):

Writing | ################################################## | 100% 6.53s

avrdude: 10562 bytes of flash written
avrdude: verifying flash memory against .pioenvs/WeatherMote_v01/firmware.hex:
avrdude: load data flash data from input file .pioenvs/WeatherMote_v01/firmware.hex:
avrdude: input file .pioenvs/WeatherMote_v01/firmware.hex contains 10562 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 5.33s

avrdude: verifying ...
avrdude: 10562 bytes of flash verified

avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as DF
avrdude: safemode: efuse reads as 7
avrdude: safemode: Fuses OK (E:07, H:DF, L:FF)

avrdude done.  Thank you.
```

ivankravets, 2015-06-01:
Thanks a lot!

Could you try new ``develop``? (2.0.3.dev2)

ncolomer, 2015-06-02:
`2.0.3.dev2` did the trick! Following is the `platformio run -t upload` output:

```
"/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude" -v -p atmega328p -C "/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude.conf" -c usbasp-clone -U flash:w:.pioenvs/WeatherMote_v01/firmware.hex:i

avrdude: Version 6.0.1, compiled on Dec 16 2013 at 17:26:24
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2009 Joerg Wunsch

System wide configuration file is "/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude.conf"

Using Port                    : usb
Using Programmer              : usbasp-clone
AVR Part                      : ATmega328P
Chip Erase delay              : 9000 us
PAGEL                         : PD7
BS2                           : PC2
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 200
StabDelay                     : 100
CmdexeDelay                   : 25
SyncLoops                     : 32
ByteDelay                     : 0
PollIndex                     : 3
PollValue                     : 0x53
Memory Detail                 :

Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

Programmer Type : usbasp
Description     : Any usbasp clone with correct VID/PID

avrdude: auto set sck period (because given equals null)
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: error: programm enable: target doesn't answer. 1
avrdude: initialization failed, rc=-1
Double check connections and try again, or use -F to override
this check.

avrdude done.  Thank you.

scons: *** [upload] Error 1
======================================================================================= [ ERROR ] Took 3.25 seconds =======================================================================================
╭─ncolomer@v581  ~/Developer/Workspace/electronics/HomeMote/WeatherMote  ‹master*›
╰─$ platformio run                                                                                                                                                                                     1 ↵
[Tue Jun  2 19:47:42 2015] Processing WeatherMote_v01 (upload_protocol: usbasp-clone, build_flags: -Llibs -DDEBUG -Wl,-u,vfprintf -lprintf_flt, install_libs: 131,132,137, platform: atmelavr, board: uno, framework: arduino, targets: upload)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/WeatherMote_v01/firmware.hex"])
"/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude" -v -p atmega328p -C "/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude.conf" -c usbasp-clone -U flash:w:.pioenvs/WeatherMote_v01/firmware.hex:i

avrdude: Version 6.0.1, compiled on Dec 16 2013 at 17:26:24
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2009 Joerg Wunsch

System wide configuration file is "/Users/ncolomer/.platformio/packages/tool-avrdude/avrdude.conf"

Using Port                    : usb
[... 46 lines cut ...]
avrdude: auto set sck period (because given equals null)
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: reading input file ".pioenvs/WeatherMote_v01/firmware.hex"
avrdude: writing flash (10950 bytes):

Writing | ################################################## | 100% 6.96s

avrdude: 10950 bytes of flash written
avrdude: verifying flash memory against .pioenvs/WeatherMote_v01/firmware.hex:
avrdude: load data flash data from input file .pioenvs/WeatherMote_v01/firmware.hex:
avrdude: input file .pioenvs/WeatherMote_v01/firmware.hex contains 10950 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 5.53s

avrdude: verifying ...
avrdude: 10950 bytes of flash verified

avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as DF
avrdude: safemode: efuse reads as 7
avrdude: safemode: Fuses OK (H:07, E:DF, L:FF)

avrdude done.  Thank you.
```

ivankravets, 2015-06-03:
Thanks! :+1:

commit c4877e44 (2015-05-29): Fix firmware uploading using USB programmer (USBasp) for atmelavr platform // Resolve #221
closed 2015-06-03 by ivankravets