platformio/platformio-core #222: "undefined reference" linker error when compiling for Arduino DUE on BeagleBone Black

eliasse, 2015-05-29:
Hello again,

Not surprisingly, I can't manage to solve this linker error on my own. 
I get the same error if I compile in OS X, but not in Linux Mint (both 64bit).
_______________________
arm-none-eabi-g++ -o .pioenvs/autogen_due/firmware.elf -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m3 -Wl,--entry=Reset_Handler -Wl,--start-group -T /home/debian/.platformio/packages/ldscripts/sam3x8e.ld .pioenvs/autogen_due/src/TCM_CLASS.o .pioenvs/autogen_due/src/tmp_ino_to.o -L/home/debian/.platformio/packages/framework-arduinosam/variants/arduino_due_x -L.pioenvs/autogen_due -Wl,--start-group -lc -lgcc -lm .pioenvs/autogen_due/libFrameworkArduinoVariant.a .pioenvs/autogen_due/libFrameworkArduino.a -lsam_sam3x8e_gcc_rel -Wl,--end-group
.pioenvs/autogen_due/src/TCM_CLASS.o: In function `TCM_MB_COMPASS::begin(HardwareSerial*, int)':
/home/debian/Desktop/due/TCM_compass/src/TCM_CLASS.cpp:14: undefined reference to `HardwareSerial::begin(unsigned long)'
**collect2: error: ld returned 1 exit status**
scons: *** [.pioenvs/autogen_due/firmware.elf] Error 1
______________________

My code is available at **https://github.com/eliasse/TCM_compass.git**

Thanks in advance!

Best Regards,
Elias

ivankravets, 2015-05-29:
Temporary solution is to replace these lines 29-30 in ~/.platformio/packages/framework-arduinosam/cores/arduino/HardwareSerial.h` by 
```c
    virtual void begin(unsigned long);
    virtual void end();
```

Need to add ``virtual`` keywords.

P.S: We are working on this issue. If we return ``virtual``, then you will have problem under ARMv(6,7).

eliasse, 2015-06-01:
I see!
I'll stay updated. I'm working on an autonomous surface vehicle project that will use an Arduino DUE
as autopilot and a BeagleBone or RPi as intelligence, so it would be very convenient to have an easy way of reprogramming the DUE remotely. 

Great work :)

valeros, 2015-06-02:
Hi @eliasse!

I updated toolchain for `atmelsam` platform, but only ARMv7 version, so your project should work on BeagleBone board.

eliasse, 2015-07-14:
Everything is working :)

Thanks again for an invaluable tool!

ivankravets, 2015-07-14:
> Thanks again for an invaluable tool!

Thanks you too for using and reporting! :+1:

closed 2015-07-14 by ivankravets