platformio/platformio-core #224: Add option to install library via git url.

jottr, 2015-05-30:
I just ran into an issue where I found that the library I need for my project is not part of the library repository. 
Since many libraries are available via github, it would be quite convenient if we could simply install those libraries like so: `platformio install https://github.com/path/to/library`
Updating libraries would be more concise as well, since we won't see that many manually installed libraries.

ivankravets, 2015-05-30:
Dear @jottr,

You can install manually own libraries which are not listed in [PlatformIO Library Registry](http://platformio.org/#!/lib). Just copy them to the ``lib`` directory of the project or into [global PlatformIO Library Directory](http://docs.platformio.org/en/latest/projectconf.html#lib-dir).

jottr, 2015-05-30:
Yes I know. But to my understanding those would not be updated with `platformio lib update`. 
Being able to install/update any library no matter their source with the library manager would be very convenient and would also remove mental overhead. 

I.e. something similar as to what [Bundler](http://bundler.io/git.html) does.

ivankravets, 2015-05-30:
The problem is hidden in the structure of the repository. I don't know which files should be used like "library source code". In most cases, developers like to keep more then 1 library in the same repo. Embedded projects/repos are more complicated then "gem", "npm", "pip" or etc packages, because they are already well-structured. 

Yes, I can allow you to install/update/manage libraries directly via CVS, but then I will not know how to compile them. 

Using [library.json](http://docs.platformio.org/en/latest/librarymanager/config.html) developer has ability to describe manifest for the own library and explain for other library manager which data should be classified like a "source code", "examples" and etc.

[PlatformIO Library Registry](http://platformio.org/#!/lib) contains libraries already prepared for building and cleaned from unnecessary files/scripts which can cause the problems on the build step. You can install a few libs using PlatformIO and then compare their source code with the real source code from repo.

jottr, 2015-05-30:
Very interesting. I was not aware of this fact, so thank you for explaining. So let's hope that many people start adding `library.json` to their repos.

ivankravets, 2015-05-30:
We're working on the automation scripts which generate ``library.json`` for the popular libraries. If you like PlatformIO you can help us with it. See https://github.com/platformio/platformio-libmirror/tree/master/scripts

Please contact with @gandy92

closed 2015-05-30 by ivankravets

LucasCanete, 2021-05-05:
Hello @ivankravets ,
I am using the esp-idf framework with platformio. I want to import a library from a github repository.
I have tried to add the .h and .c files from https://github.com/UncleRus/esp-idf-lib/tree/master/components/hd44780 to the lib folder as you mentioned in this discussion but when i include the .h file to main.c the program cant find the header. Could you offer me some guidance with this?

Greetings 

Lucas