platformio/platformio-core #225: Error linking: multiple definition of `__ltsf2'

lekum, 2015-05-31:
Hi,

I'm using platformio 2.0.2 on armv6 (raspberry pi) with Arch Linux installed.

I can compile and upload the basic blink sketch, but when I try with a more complex sketch like this [one](https://github.com/lekum/pyduino/blob/master/pyduino_sketch.ino) I get this error when trying to link:

```
The platform 'atmelavr' has been successfully installed!
avr-g++ -o .pioenvs/uno/firmware.elf -Os -mmcu=atmega328p -Wl,--gc-sections,--relax .pioenvs/uno/src/tmp_ino_to.o -L.pioenvs/uno -Wl,--start-group -lm .pioenvs/uno/libFrameworkArduinoVariant.a .pioenvs/uno/libFrameworkArduino.a -Wl,--end-group
/home/alex/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.7.2/../../../avr/lib/avr5/libc.a(cmpsf2.o):../../../libm/fplib/cmpsf2.S:58: multiple definition of `__ltsf2'
/home/alex/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.7.2/avr5/libgcc.a(_lt_sf.o):(.text+0x0): first defined here
/home/alex/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.7.2/../../../avr/bin/ld: Disabling relaxation: it will not work with multiple definitions
/home/alex/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.7.2/../../../avr/lib/avr5/libc.a(cmpsf2.o):../../../libm/fplib/cmpsf2.S:58: multiple definition of `__lesf2'
/home/alex/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.7.2/avr5/libgcc.a(_le_sf.o):(.text+0x0): first defined here
```

The platform target is atmealvr (for Arduino Uno).

Thanks!

valeros, 2015-06-01:
Hi @lekum!

I updated `toolchain-atmelavr` for ARMv6.
Can you test it, please?

lekum, 2015-06-01:
Now it works, with the newest version of `toolchain-atmelavr`. Thanks!

closed 2015-06-01 by lekum