platformio/platformio-core #228: Support espressif other modules too.

middelink, 2015-06-03:
Espressif allows one to add different sizes flash to a module, so being able to set the memory size _is_ pretty useful. Especially in light of the fact the current libat.a requires 1 MB flash or better. Sadly platformio has various hardcoded places where it uses 512 MB.

Also, dropping esp8266.flash.512k.ld in platformio/packages/ldscripts seems awkward as  platformio/packages/framework-arduinoespressif/sdk/ld/ already contains the "official" set of linker scripts, all the way up to 16 MB.

Also not 100% sure why the sdk (sdk-esp8266) is downloaded as well, it does not seem to be used. (Worse, the sdk is version 1.0.0 whereas the framework has sdk version 1.0.1_15_05_04_p1 - which is not the latest, btw)

ivankravets, 2015-06-04:
Dear @middelink 

You can create/duplicate own board with the specified settings. Please look into documentation: http://docs.platformio.org/en/latest/platforms/creating_board.html

Here is original version of board config: https://github.com/platformio/platformio/blob/develop/platformio/boards/espressif.json

If you create board with the same type, it will redefine built-in data for this board.

> Also not 100% sure why the sdk (sdk-esp8266) is downloaded as well, it does not seem to be used. (Worse, the sdk is version 1.0.0 whereas the framework has sdk version 1.0.1_15_05_04_p1 - which is not the latest, btw)

PlatformIO uses ``sdk-esp8266`` for native projects, without Arduino framework. On the other hand, Espressif Arduino framework uses own version of SDK which has a few modifications.

valeros, 2015-06-10:
Hi @middelink!

The easiest way is create new platform and board for your needs.
Detailed information about creating platform:
http://docs.platformio.org/en/latest/platforms/creating_platform.html
Detailed information about creating board:
http://docs.platformio.org/en/latest/platforms/creating_board.html

You can take the existing platform `espressif`  and board `esp01` with minimal changes:
Modify board memory size in the upload command: https://github.com/platformio/platformio/blob/develop/platformio/builder/scripts/espressif.py#L95
Specify path to your `ldscript`:
https://github.com/platformio/platformio/blob/develop/platformio/boards/espressif.json#L7

It should be enough.

valeros, 2015-06-10:
Please reopen this issue if you still have problems.

closed 2015-06-10 by valeros