platformio/platformio-core #229: Avoid "LibInstallDependencyError" when more then 1 library is found

ivankravets, 2015-06-06:
Need to ask user via prompt which dependent lib should be installed.

closed 2015-06-06 by ivankravets

pelikhan, 2015-06-06:
No prompts!

ivankravets, 2015-06-06:
@pelikhan Maybe, this is the most annoying feature in PlatformIO for you :smile: These prompts...

Nevertheless, please look [into code](https://github.com/platformio/platformio/commit/3b7ad2d3338c5a1cb5aa82198829be499ef4c512#diff-86ad8f7e8aafa191f30dab69b84f4736R147):
```python
    result = get_api_result("/lib/search", dict(query=" ".join(query)))
    assert result['total'] > 0

    if result['total'] == 1 or not app.get_setting("enable_prompts"):
        ctx.invoke(lib_install, libid=[result['items'][0]['id']])
    else:
        click.secho(
            "Conflict: More then one dependent libraries have been found "
            "by request %s:" % json.dumps(data), fg="red")

        echo_liblist_header()
        for item in result['items']:
            echo_liblist_item(item)

        deplib_id = click.prompt(
            "Please choose one dependent library ID",
            type=click.Choice([str(i['id']) for i in result['items']]))
        ctx.invoke(lib_install, libid=[int(deplib_id)])
```

If you can see, the prompts wont be used if 1 dependent library was found. If more then 1(this is conflict), then I have to ask user which dependent library (a list with dependent candidates will be shown) need to install. 

Nevertheless, since PlatformIO 2.0 you can use ``-f/--force`` option before each command which automatically disable all prompts. For example, to install library with CONFLICTED dependencies and avoid prompts use:
```shell
platformio --force lib install %ID%
```

In this situation, PlatformIO will install the first library from the search result which is matched by dependent request.

pelikhan, 2015-06-06:
In doubt, you should fail if --force is used.

commit 3b7ad2d3 (2015-06-06): Avoid "LibInstallDependencyError" when more then 1 library is found // Resolve #229