platformio/platformio-core #232: Need to press RESET button to upload

venusshum, 2015-06-08:
>platformio run

E:\01-Development\SVN\Engduino-3rd-party\MatlabSupport\Protocol>platformio run
[06/08/15 15:48:31] Processing engduinov3 (targets: upload, platform: atmelavr,
board: engduinov3, framework: arduino)
-------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs\engduinov3\firmware.hex"])
Auto-detected UPLOAD_PORT: COM4
scons: *** [upload] Explicit exit, status Error: Couldn't find a board on the se
lected port. Check that you have the correct port selected. If it is correct, tr
y pressing the board's reset button after initiating the upload.
Error: Couldn't find a board on the selected port. Check that you have the corre
ct port selected. If it is correct, try pressing the board's reset button after
initiating the upload.
======================== [ ERROR ] Took 12.30 seconds ========================

> press reset twice  on the board after "platformio run"
=> success

ivankravets, 2015-06-08:
I hope that this issue was fixed in https://github.com/platformio/platformio/commit/f868c64d819a660791fc5e14862b0dee7d8c219e.

Please re-test it in ``develop`` version of PlatformIO: http://docs.platformio.org/en/latest/installation.html#development-version

Did it help you?

venusshum, 2015-06-08:
It works. Thanks ☺

Would I be notified when it moves to the release version please?

I hope that this issue was fixed in f868c64<https://github.com/platformio/platformio/commit/f868c64d819a660791fc5e14862b0dee7d8c219e>.

ivankravets, 2015-06-08:
I'm going to release PlatformIO 2.1.1 tomorrow and report here (you will be notified) :smile: 

Thanks for the report! :+1:

closed 2015-06-08 by ivankravets

ivankravets, 2015-06-10:
Dear @venusshum ,

PlatformIO 2.1.1 is released! https://twitter.com/platformio_org/status/608567598477942784

venusshum, 2015-06-23:
Hi, Ivan,

Do you have any documentation how platformio can be implemented as a web service please?

Many thanks.
Venus

ivankravets, 2015-06-23:
> Do you have any documentation how platformio can be implemented as a web service please?

What do you mean under web-service? If you are going to call platformio from subprocess, don't forget to use ``--force or -f`` option before each command. See details [here](http://docs.platformio.org/en/latest/faq.html#platformio-blocks-command-execution-using-user-prompt).

For example,
```shell
platformio -f run
platformio -f ci
```