platformio/platformio-core #233: Allow to specify own path to linker script (ld) using "build_flags" option

ivankravets, 2015-06-10:
Process ``-Wl,-T%PATH_TO_LD_SCRIPT%`` option from http://docs.platformio.org/en/latest/projectconf.html#build-flags

me21, 2019-07-18:
Correct link is now http://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-flags

me21, 2019-07-18:
@ivankravets I try to pass this option to linker:

```
build_flags = -DARDUINO_BUILD -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -Wl,-T${platformio.src_dir}\flash.ld,-v
```

but all I get now is
```
Linking .pio\build\bluepill_f103c8_arduino\firmware.elf
.pio\build\bluepill_f103c8_arduino\FrameworkArduino\main.cpp.o: In function `main':
main.cpp:(.text.startup.main+0x2): warning: undefined reference to `setup()'
main.cpp:(.text.startup.main+0x6): warning: undefined reference to `loop()'
```

ivankravets, 2019-07-18:
> build_flags = -Wl,-T${platformio.src_dir}\flash.ld,-

It should be `build_flags = -Wl,-T$PROJECT_DIR/src/flash.ld,-v`

http://docs.platformio.org/en/latest/projectconf/section_env_build.html#built-in-variables

me21, 2019-07-18:
@ivankravets Went a little further by using `-Wl,-v,-T"C:\Temp\Platformio Projects\bluepill\src\flash.ld"`, but now -T option is included in linker command line twice, once from `build_flags`, once from the build script.

```
Linking .pio\build\bluepill_f103c8_arduino\firmware.elf
collect2 version 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
c:/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe -plugin c:/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/liblto_plugin-0.dll -plugin-opt=c:/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/lto-wrapper.exe -plugin-opt=-fresolution=C:\Users\ED71~1\AppData\Local\Temp\ccYommBf.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc --sysroot=c:\.platformio\packages\toolchain-gccarmnoneeabi@1.70201.0\bin\../arm-none-eabi -X -o .pio\build\bluepill_f103c8_arduino\firmware.elf c:/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/thumb/v7-m/crti.o c:/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/thumb/v7-m/crtbegin.o c:/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/crt0.o -LC:\.platformio\platforms\ststm32\ldscripts -L.pio\build\bluepill_f103c8_arduino -LC:\.platformio\packages\framework-arduinoststm32-maple\STM32F1\variants\maple\ld -Lc:/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/thumb/v7-m -Lc:/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7-m -Lc:/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../arm-none-eabi/lib/thumb/v7-m -Lc:/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1 -Lc:/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc -Lc:/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib -Lc:/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../arm-none-eabi/lib -v -TC:\Temp\Platformio Projects\bluepill\src\flash.ld --check-sections --gc-sections --unresolved-symbols=report-all --warn-common --warn-section-align --warn-unresolved-symbols .pio\build\bluepill_f103c8_arduino\FrameworkArduinoVariant\board.cpp.o .pio\build\bluepill_f103c8_arduino\FrameworkArduinoVariant\wirish\boards.cpp.o .pio\build\bluepill_f103c8_arduino\FrameworkArduinoVariant\wirish\boards_setup.cpp.o .pio\build\bluepill_f103c8_arduino\FrameworkArduinoVariant\wirish\start.S.o .pio\build\bluepill_f103c8_arduino\FrameworkArduinoVariant\wirish\start_c.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduinoVariant\wirish\syscalls.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\HardwareSerial.cpp.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\HardwareTimer.cpp.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\IPAddress.cpp.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\Print.cpp.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\Stream.cpp.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\WString.cpp.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\avr\dtostrf.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\cxxabi-compat.cpp.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\ext_interrupts.cpp.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\hooks.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\itoa.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\adc.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\adc_f1.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\bkp_f1.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\dac.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\dma.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\dma_f1.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\exc.S.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\exti.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\exti_f1.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\flash.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\fsmc_f1.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\gpio.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\gpio_f1.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\i2c.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\i2c_f1.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\iwdg.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\nvic.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\pwr.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\rcc.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\rcc_f1.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\spi.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\spi_f1.c.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\stm32f1\performance\isrs.S.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\stm32f1\performance\vector_table.S.o .pio\build\bluepill_f103c8_arduino\FrameworkArduino\libmaple\systick.c.o .pio\build\bluep [... cut ...]
c:/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe:mem-flash.inc:4: warning: redeclaration of memory region `rom'
c:/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe:flash.ld:20: error: redefinition of memory region alias `REGION_TEXT'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\bluepill_f103c8_arduino\firmware.elf] Error 1
```

ivankravets, 2019-07-18:
If you want to pass complex flags I recommend to do this directly with build environment using an extra script (PRE). See https://github.com/platformio/platformio-core/blob/develop/platformio/builder/tools/platformio.py#L127

me21, 2019-07-18:
So specifying own path to linker script in `build_flags` is not supported now?

me21, 2019-07-18:
@ivankravets The combination of `build_flags` which works for me is 
```
build_flags = -Wl,-v -Wl,-T"C:\Temp\Platformio Projects\bluepill\src\flash.ld"
```
i.e. specify two separate flags to be passed to linker. 
And since my project path contains spaces, your proposal `build_flags = -Wl,-T$PROJECT_DIR/src/flash.ld,-v` didn't work for me.

But there's another bug: linker script includes other files. Example: `INCLUDE mem-flash.inc`. I placed `mem-flash.inc` next to my custom linker file, but apparently `mem-flash.inc` from framework's `variants/maple/ld/` folder is preferred. Possibly if custom linker file is specified, it should include files from its containing folder first.

me21, 2019-07-18:
Okay, I found how to escape spaces in path: `-Wl,-T\"$PROJECT_DIR/src/flash.ld\",-v`. But the other problem with linker script including other files from framework subdirectory first still holds.

It's not a big problem, after all, since I can now edit my custom linker script and rename included files. But it's unexpected, at least for me.

me21, 2019-07-18:
Okay, I've found how to escape spaces in path: `-Wl,-T'"$PROJECT_DIR\src\flash.ld"',-v`.

And to make the linker include my changed files, I need to specify their full path in linker script.

Here are my linker script and include file for reference:

Linker script `flash.ld`:
```
/*
 * libmaple linker script for "Flash" builds.
 *
 * A Flash build puts .text (and .rodata) in Flash, and
 * .data/.bss/heap (of course) in SRAM, but offsets the sections by
 * enough space to store the Maple bootloader, which lives in low
 * Flash and uses low memory.
 */

/*
 * This pulls in the appropriate MEMORY declaration from the right
 * subdirectory of stm32/mem/ (the environment must call ld with the
 * right include directory flags to make this happen). Boards can also
 * use this file to use any of libmaple's memory-related hooks (like
 * where the heap should live).
 */
INCLUDE "C:\Temp\Platformio Projects\bluepill\src\mem-flash.inc"

/* Provide memory region aliases for common.inc */
REGION_ALIAS("REGION_TEXT", rom);
REGION_ALIAS("REGION_DATA", ram);
REGION_ALIAS("REGION_BSS", ram);
REGION_ALIAS("REGION_RODATA", rom);

/* Let common.inc handle the real work. */
INCLUDE common.inc
```

Included file `mem-flash.inc`:
```
MEMORY
{
  ram (rwx) : ORIGIN = 0x20000C00, LENGTH = 17K
  rom (rx)  : ORIGIN = 0x08002000, LENGTH = 120K
}
```

ivankravets, 2019-07-19:
> Okay, I've found how to escape spaces in path: -Wl,-T'"$PROJECT_DIR\src\flash.ld"',-v.

This is correct format. Thanks for the example

closed 2019-07-19 by ivankravets

me21, 2019-07-19:
@ivankravets, just for the sake of completeness - when I include other files on linker scripts, do I have to specify full path as I did? Specifying filename only didn't work for me.

ivankravets, 2019-07-19:
The CWD is current project directory.

jamesle-0812, 2020-01-07:
Hi there, I'm new with Platformio. Recently I use a custom bootloader to upload to STM32F103, based on libopencm3 framework. However, the bootloader author said I have to change the ROM address from default to 0x08002000. Is there any possible way to configure it in platformio.ini? I don't wanna mess file system up!

commit 8e95bfb4 (2015-06-19): Allow to specify own path to the linker script (ld) using build_flags option // Resolve #233