platformio/platformio-core #234: Hard-to-read colors in console on Winows

divanchykhin, 2015-06-11:
More particularly, using cmd.exe on Windows 7 the following colors appeared: 
[image]
(this might or might not be a problem specific to a particular installation)

ngollon, 2015-09-03:
The dark red is especially bad. 

It helps  a little to set your console's background to white and text color to black in the console's properties.

ivankravets, 2015-09-04:
This issue isn't linked with PlatformIO. You should customise your Terminal. It uses "grayed" colors scheme by default. Click on the `Terminal icon(left top corner of window) -> Properties -> Colors -> Screen Text` and set to "white" (255,255,255). See
[image]

closed 2015-09-04 by ivankravets