platformio/platformio-core #236: Plugins for setting build options based on EEPROM contents?

aphelps, 2015-06-13:
This is a request that may be pretty specific to my needs, but I figured I'd ask.

The devices I'm working on use a single board but may have a number of different configurations of hardware attached to them, and so I use compile switches to specify which libraries and such need to be compiled in, and I store the configuration in EEPROM.

Rather than needing a different environment in the platformio.ini file for each configuration I just started on a script to read and parse the EEPROM and then generate the appropriate compile flags.  I figured rather than keeping it specific to my usage, I figured I'd ask here about getting a more generic method of running a plugin that could modify the build environment.

For this to work the plugin would need access to the environment object, the platform object, and have some method of getting the appropriate command for the platform.  For instance in my case I needed this from the output of 'platformio run' as a base for dumping the EEPROM contents:

```
"/Users/amp/.platformio/packages/tool-avrdude/avrdude" -v -D -p atmega328p -C "/Users/amp/.platformio/packages/tool-avrdude/avrdude.conf" -c arduino -b 57600 -P /dev/cu.usbserial-A9015AMZ
```

I put together the basics of what I think would be needed for this here: https://github.com/aphelps/platformio/commit/27c2a505d3361da1bd93578274b58c5be6a34b2d

The thing I couldn't quite figure out is how to get the above command out of the BasePlatform object.  If you folks could provide pointers there I might be able to just finish this one off myself.

ivankravets, 2015-06-14:
How about new option in ``[env:***]`` group name ``extra_script``? Here you can pass the path to own build script and make manipulation over building environment using [SCons](http://www.scons.org). See examples: https://github.com/platformio/platformio/tree/develop/platformio/builder/scripts

aphelps, 2015-06-15:
That could work.  Are those scripts ultimately what's called to do the building and uploading?  It looks like it to me as that's where avrdude actually seems to be called.

Looking at this I'd basically need to produce a modified version of atmelavr.py, and then the new extra_script option would be used to call my version rather than the built in one?  Or would it call extra_script first and then call the built in build script?

ivankravets, 2015-06-15:
This can be replacement for " I'd basically need to produce a modified version of atmelavr.py". This is script will be called before firmware building, in this line: https://github.com/platformio/platformio/blob/develop/platformio/builder/tools/platformio.py#L61

It allows you to perform any changes on build environment and manipulate over uploading command and flags. To dump current environment please use ``platformio run --target envdump``. You will see all options/flags which will be use for firmware building and uploading.

Later using ``extra_script`` you can correct any of these environment variables. For example, ``myscript.py``:
```python
from SCons.Script import DefaultEnvironment

env = DefaultEnvironment()

# print current state of environment
# print env.Dump()

env.Replace(
    UPLOADERFLAGS="my own uploader flags"
)

env.Append(
    CPPDEFINES=["DEF1", "DEF2=123"]
)
```

See documentation: http://www.scons.org/doc/production/HTML/scons-user.html#chap-environments

ivankravets, 2015-06-19:
@aphelps Could I ask you to try new [development version of PlatformIO](http://docs.platformio.org/en/latest/installation.html#development-version) with ``extra_script`` feature? See #239.

ivankravets, 2015-09-07:
Please reopen this issue if you still need help. `extra_script` sounds like a good candidate. Then you can call different(own) scripts and apply specific flags.

closed 2015-09-07 by ivankravets