platformio/platformio-core #237: RaspberryPi 2 (Raspbian) - atmelavr - Build Fails

ctag, 2015-06-13:
Hi,

I just learned about PlatformIO and I'm trying to use it to bridge the gap between my desktop and Raspberry Pi for an ATmega project. On my desktop the package installs and works brilliantly. It also installs on the RPi, but when I go to build the same project, I receive a build error.

Installing:
```
$ python --version
Python 2.7.3
$ sudo python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"
[...]
$ platformio --version
PlatformIO, version 2.1.1
```

Building Project:
```
$ platformio run
[Fri Jun 12 22:16:23 2015] Processing diecimilaatmega328 (platform: atmelavr, board: diecimilaatmega328, framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------
avr-g++ -o .pioenvs/diecimilaatmega328/src/tmp_ino_to.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_DUEMILANOVE -DARDUINO=10604 -DPLATFORMIO=020101 -I.pioenvs/diecimilaatmega328/FrameworkArduino -I.pioenvs/diecimilaatmega328/FrameworkArduinoVariant -I.pioenvs/diecimilaatmega328/Servo -I.pioenvs/diecimilaatmega328/amt1001 src/tmp_ino_to.cpp
/home/pi/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.8.1/cc1plus: error while loading shared libraries: libmpc.so.3: cannot open shared object file: No such file or directory
scons: *** [.pioenvs/diecimilaatmega328/src/tmp_ino_to.o] Error 1
============================================================= [ ERROR ] Took 4.52 seconds =============================================================
```
platformio.ini
```
[env:diecimilaatmega328]
platform = atmelavr
framework = arduino
board = diecimilaatmega328
# targets = upload
upload_port = /dev/ttyAMA0
```
I've also tried creating a virtualenv and installing platformio+scons with pip, but wind up with the same compiling error as above.
Please let me know if I can gather any more information to help with debugging this.

Thanks!,
-ctag

valeros, 2015-06-13:
Hi @ctag!
Please, try to install mpc library:
`sudo apt-get install libmpc-dev`

ctag, 2015-06-13:
@valeros,

After installing the `libmpc-dev` and `libelf1`, PlatformIO builds and uploads the project. Thank you!

closed 2015-06-13 by ctag