platformio/platformio-core #238: stk500v2_command(): command failed

mpd-dude, 2015-06-16:
Hi all,

when I perform a "platformio run --target upload"

the following error messages appeards:
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
To disable this feature, specify the -D option.
__avrdude: erasing chip__
__avrdude: stk500v2_command(): command failed__
avrdude done.  Thank you.
scons: *** [upload] Error 255

Inside my "platformio.ini" file I have
[env:autogen_megaADK]
platform = atmelavr
framework = arduino
board = megaatmega2560

I am using an Arduino MEGA 2560 under Win7.

A simple "platformio run" runs without any problems and compilation seems ok.

The same sketch can be uploaded using the Arduino IDE without any problems.

I am a bit stuck here. Any hints are appreceated.

Thanks,
Lars

ivankravets, 2015-06-17:
Thanks for the report!

Please try development version http://docs.platformio.org/en/latest/installation.html#development-version

Did it help you?

closed 2015-06-17 by ivankravets

mpd-dude, 2015-06-17:
Thanks a lot for the quick reply and the fix!

Yes, it fixed the problem and uploading is working again.

commit 4669dc7f (2015-06-17): Fix "stk500v2_command(): command failed" // Resolve #238