platformio/platformio-core #241: --board=teensy31 fails on my Chromebook

hoosierEE, 2015-06-21:
## Here's what I did:

1. `mkdir foo && cd foo`
2. `platformio init --board=teensy31` (answer prompts with `n`, then `y`)
3. write test file `src/src.ino` which contains:

   ```
   void setup(){}
   void loop() { int i = 0; }
   ```
4. `platformio run`

## Here's what I get

```
[Sun Jun 21 13:18:26 2015] Processing teensy31 (platform: teensy, board: teensy31, framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------
arm-none-eabi-g++ -o .pioenvs/teensy31/src/tmp_ino_to.o -c -fno-rtti -fno-exceptions -std=gnu++0x -felide-constructors -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -MMD -DF_CPU=72000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -D__MK20DX256__ -DARDUINO=10600 -DTEENSYDUINO=121 -DPLATFORMIO=020101 -I.pioenvs/teensy31/FrameworkArduino src/tmp_ino_to.cpp
/home/lx/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.3/cc1plus: error while loading shared libraries: libmpc.so.2: cannot open shared object file: No such file or directory
scons: *** [.pioenvs/teensy31/src/tmp_ino_to.o] Error 1
============================================================ [ ERROR ] Took 1.07 seconds ============================================================
```

## More info

The above steps run with **SUCCESS** if I change `teensy31` for `nanoatmega328`.  My laptop is a Samsung 1st gen ARM Chromebook with Ubuntu 14.04.

```
$ uname -a
Linux localhost 3.8.11 #1 SMP Sat Jun 13 02:36:36 PDT 2015 armv7l armv7l armv7l GNU/Linux
```

ivankravets, 2015-06-21:
You need to install ``libmpc-dev`` package for Ubuntu:
```shell
apt-get install libmpc-dev
```

hoosierEE, 2015-06-21:
No change after trying your suggestion.

ivankravets, 2015-06-21:
Can you try solution from here? http://stackoverflow.com/questions/19625451/cc1-error-while-loading-shared-libraries-libmpc-so-2-cannot-open-shared-objec

hoosierEE, 2015-06-21:
Tried all solutions listed, same result.  Thanks!

ivankravets, 2015-06-21:
@valeros, please help @hoosierEE

valeros, 2015-06-22:
Hi @hoosierEE !
Can you try to install only libmpc2:
`sudo apt-get install libmpc2`

hoosierEE, 2015-06-22:
I guess not: 

```
$ sudo apt-get install libmpc2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libmpc2
```

Here's what I do have on `libmpc`:
```
$ apt-cache search libmpc
libmpc-dev - multiple precision complex floating-point library development package
libmpc3 - multiple precision complex floating-point library
libmpcdec-dev - MusePack decoder - development files
libmpcdec6 - MusePack decoder - library
mppenc - Musepack lossy audio codec encoder
```

`libmpc3` and `libmpc-dev` are already installed on my system.

valeros, 2015-06-22:
Do you have gcc package on your notebook?

hoosierEE, 2015-06-22:
Yep:
```
$ gcc --version
gcc (Ubuntu/Linaro 4.8.2-19ubuntu1) 4.8.2
```

valeros, 2015-06-22:
Can you try to install this package:
http://ftp.de.debian.org/debian/pool/main/m/mpclib/libmpc2_0.9-4_armel.deb
`sudo dpkg -i libmpc2_0.9-4_armel.deb`

hoosierEE, 2015-06-22:
Here's what I get
```
$ sudo dpkg -i libmpc2_0.9-4_armel.deb
dpkg: error processing archive libmpc2_0.9-4_armel.deb (--install):
 cannot access archive: No such file or directory
Errors were encountered while processing:
 libmpc2_0.9-4_armel.deb
```

hoosierEE, 2015-06-22:
Would it be possible to use libmpc3?

valeros, 2015-06-22:
Please, try the complete procedure:
```
mkdir /tmp/mpc && cd /tmp/mpc
wget http://ftp.de.debian.org/debian/pool/main/m/mpclib/libmpc2_0.9-4+b1_armhf.deb
sudo dpkg -i libmpc2_0.9-4+b1_armhf.deb
sudo ldconfig
```
This approach works fine on my cubieboard (armv7)

hoosierEE, 2015-06-22:
`======== [SUCCESS] Took 12.68 seconds =========`

That did it!  Thanks a bunch!

hoosierEE, 2015-06-22:
Also, I want to add that the response from collaborators on this issue has been amazing, you all should be proud of yourselves. :+1:

closed 2015-06-22 by hoosierEE

ivankravets, 2015-06-22:
Thanks you too! I've just added this issue to FAQ: http://docs.platformio.org/en/latest/faq.html#arm-toolchain-cc1plus-error-while-loading-shared-libraries