platformio/platformio-core #242: IDE generator only builds project for last env in existing platformio.ini file

pastcompute, 2015-06-25:
When I run the following command as suggested by the tutorial:
```
platformio init --ide qtcreator --board esp01
```
and the `platformio.ini` file in the directory contains a second `env:teensylc` section after the `env:esp01` section, the generator ignored the `--board iesp0` option and always generates a teensylc-based qtcreator project.

I had a browse through the code and noted in `projectgenerator.py` it scans through the ini file and populates a map in function `get_project_env()` and this seems to take no notice of the `ide` selection;  for any number of `env` sections they will keep getting replaced (thus the choice of the last in the file)
I guess a fix would involve reworking how that method gets called, maybe a second loop in generate() over boards, but my head is not in Python mode at the moment

ivankravets, 2015-06-25:
Thanks, I'll fix it in 2.2.0 release.

closed 2015-06-27 by ivankravets

ivankravets, 2015-06-27:
Please re-test it using [development version](http://docs.platformio.org/en/latest/installation.html#development-version).

commit 82864d38 (2015-06-27): Fix IDE project generator when board is specified // Resolve #242