platformio/platformio-core #243: Generated QtCreator project contains hard-coded paths to $HOME/.platformio directory

pastcompute, 2015-06-25:
The project file generated by running `platformio init --ide qtcreator` uses absolute paths to the platformio installed SDKs for `INCLUDE` paths, etc.

For example:
`INCLUDEPATH += "/home/pastcompute/.platformio/packages/framework-arduinoespressif/variants/generic"`
This makes it unsuitable for adding to version control.
It would be prefereable if these options were instead added like:
`INCLUDEPATH += "${HOME}/.platformio/packages/framework-arduinoespressif/variants/generic"`
making the project portable between users

ivankravets, 2015-06-27:
Please re-test it using [development version](http://docs.platformio.org/en/latest/installation.html#development-version).

Does it work for you?

closed 2015-06-27 by ivankravets

pastcompute, 2015-06-30:
Hi, I just tested this and it works. Thanks!

commit ae4918bd (2015-06-27): Fix relative path for includes when generating project for IDE // Resolve #243