platformio/platformio-core #245: ESP8266 native SDK expection

condemil, 2015-06-27:
ESP8266 package with native SDK gives the following exception:

    Fatal exception (0): epc1=0x40240655, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

With arduino framework ESP8266 is working fine, I can see such behaviour only with native SDK

How to reproduce

* Create native SDK project with your `upload_port` in `platformio.ini`

*. Create empty `user_config.h`  under the src folder

* Create `user_main.c` under the src folder with the following content:

```c
#include "ets_sys.h"
#include "osapi.h"
    
void user_init(void)
{
    os_printf("SDK version:%s\n", system_get_sdk_version());
}
```

* Execute `platformio run`

* Connect to ESP8266 UART with 74880 baudrate, I am using `miniterm` from `pyserial` for that purpose

```shell
miniterm.py /dev/<YOUR_DEV> -b 74880
```

ivankravets, 2015-06-27:
> miniterm.py /dev/<YOUR_DEV> -b 74880

Don't forget that you can use [platformio serialports monitor](http://docs.platformio.org/en/latest/userguide/cmd_serialports.html#platformio-serialports-monitor) command. It also uses ``miniterm``.

closed 2015-06-30 by valeros

ivankravets, 2015-06-30:
@condemil please install [development version of PlatformIO](http://docs.platformio.org/en/latest/installation.html#development-version) and update packages:
```shell
> platformio --version
# PlatformIO, version 2.2.0b1

> platformio update
```

Has it helped you?

commit c80fb5f4 (2015-06-30): Update Espressif platform // Resolve #245