platformio/platformio-core #247: avrdude verification error when uploading to Panstamp AVR

gandy92, 2015-07-02:
I am working on a firmware that makes use almost all of Panstamp AVR resources. I have a working set of features that compiles with the result
```
AVR Memory Usage
----------------
Device: atmega328p

Program:   27574 bytes (84.1% Full)
(.text + .data + .bootloader)

Data:       1370 bytes (66.9% Full)
(.data + .bss + .noinit)

EEPROM:       15 bytes (1.5% Full)
(.eeprom)
```
and uploads with the message:
```
Using Port                    : /dev/ttyUSB0
Using Programmer              : arduino
Overriding Baud Rate          : 57600
AVR Part                      : ATmega328P
Chip Erase delay              : 9000 us
PAGEL                         : PD7
BS2                           : PC2
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 200
StabDelay                     : 100
CmdexeDelay                   : 25
SyncLoops                     : 32
ByteDelay                     : 0
PollIndex                     : 3
PollValue                     : 0x53
Memory Detail                 :

Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

Programmer Type : Arduino
Description     : Arduino
Hardware Version: 2
Firmware Version: 1.16
Vtarget         : 0.0 V
Varef           : 0.0 V
Oscillator      : Off
SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f
avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: reading input file ".pioenvs/asksin/firmware.hex"
avrdude: writing flash (27574 bytes):

Writing | ################################################## | 100% 7.71s

avrdude: 27574 bytes of flash written
avrdude: verifying flash memory against .pioenvs/asksin/firmware.hex:
avrdude: load data flash data from input file .pioenvs/asksin/firmware.hex:
avrdude: input file .pioenvs/asksin/firmware.hex contains 27574 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 5.63s

avrdude: verifying ...
avrdude: 27574 bytes of flash verified

avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (H:00, E:00, L:00)

avrdude done.  Thank you.
```
This firmware successfully runs on the Panstamp AVR board.

However, when I compile in one more feature, I get:
```
AVR Memory Usage
----------------
Device: atmega328p

Program:   31292 bytes (95.5% Full)
(.text + .data + .bootloader)

Data:       1611 bytes (78.7% Full)
(.data + .bss + .noinit)

EEPROM:       15 bytes (1.5% Full)
(.eeprom)
```
which IMHO suggests that it should fit in the Panstamp memory. 

However, when I try to upload this, I get:
```
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f
avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: reading input file ".pioenvs/asksin/firmware.hex"
avrdude: writing flash (31292 bytes):

Writing | ################################################## | 100% 8.55s

avrdude: 31292 bytes of flash written
avrdude: verifying flash memory against .pioenvs/asksin/firmware.hex:
avrdude: load data flash data from input file .pioenvs/asksin/firmware.hex:
avrdude: input file .pioenvs/asksin/firmware.hex contains 31292 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 6.42s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x7000
0xff != 0xef
avrdude: verification error; content mismatch

avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (H:00, E:00, L:00)

avrdude done.  Thank you.

scons: *** [upload] Error 1
```

This result does not depend on the additional feature per se, I can deactivate other features and get working uploads as long as I do not exceed a certain program or data memory size. Every time the upload fails, the error occurs at byte 0x7000. 
Is it possible the memory sizes are not correctly configured for the Panstamp AVR board? Or is there something wrong with the upload process?
I was having this problem for some time now and just updated to PIO 2.2.0 to verify the problem is still there.

ivankravets, 2015-07-02:
Could I ask you to try [extra_script](http://docs.platformio.org/en/latest/projectconf.html#extra-script) with this content:
```python
from os.path import join
from SCons.Script import DefaultEnvironment

env = DefaultEnvironment()

env.Replace(
    MYUPLOADERFLAGS=[
        "-v",
        "-p", "$BOARD_MCU",
        "-C",
        '"%s"' % join("$PIOPACKAGES_DIR", "tool-avrdude", "avrdude.conf"),
        "-c", "$UPLOAD_PROTOCOL",
        "-b", "$UPLOAD_SPEED"
     ],
    UPLOADHEXCMD='"$UPLOADER" $MYUPLOADERFLAGS -U flash:w:$SOURCES:i'
)
```

Has it helped you?

gandy92, 2015-07-02:
Wow, that was fast, thank you :smile: 

I had to add  `"-P", "$UPLOAD_PORT"` to make it use ttyUSB0 instead of default ttyS0, but again I get the verification error at byte 0x7000.

ivankravets, 2015-07-03:
Have you tried this project with original Arduino IDE with panStamp core? See https://github.com/panStamp/panstamp/wiki/Installing-panStamp-cores-and-libraries-for-Arduino

gandy92, 2015-07-08:
Thanks for the suggestion. Meanwhile, I managed to install the IDE and get
the firmware compiling which required quite some changes to source files
(which in fact helped reminding me several times why I prefer PlatformIO so
much over the arduino IDE). In the end, the code compiled and the binary
file was even larger compared to that compiled with platformio (Even though
I'm quite positive I'm compiling in the same set of features). There is no
error output regarding the size of the binary but the IDE does state a
warning that due to the large size, stability problems are to be expected.

I didn't upload this to the panstamp yet to check the upload process and if
it is running afterwards, and it may take some more time before I can get
my hands on the hardware and do so. But I'll keep you updated.

What I did learn trying to find out about related problems and possible
solutions, it that there are various bootloaders around with varying size
and that the bootloader lives in the upper memory region. Maybe in my
panstamp it starts at address 0x7000? Which would mean that the topmost 4k
are reserved for the bootloader? I'll have to read some more.

ivankravets, 2015-07-14:
Have you resolved this issue?

gandy92, 2015-07-17:
I just hooked up the board to arduino IDE and I get the very same error
```
avrdude: verification error, first mismatch at byte 0x7000
         0xff != 0xbe
avrdude: verification error; content mismatch
```
So clearly it is not a problem of PIO per se. 

Thank you for your support in figuring this out!

closed 2015-07-17 by gandy92

ivankravets, 2015-07-17:
@gandy92 thanks for the report! We've released 2.2.1 today :smile: Don't forget to upgrade.

gandy92, 2015-07-17:
Thanks for the heads up, I will update and try it out right after having removed that other IDE :wink: