platformio/platformio-core #248: Setting baud requires setting port

efatsi, 2015-07-07:
Getting this error when I try to just set the `baud` option
```
$ platformio serialports monitor --baud 115200
Usage: --baud [options] [port [baudrate]]

--baud: error: no arguments are allowed, options only when --port is given
```

Works if I set the `port` option as well (as long as the port is set before baud)
```
& platformio serialports monitor --port /dev/cu.usbmodem1421 --baud 115200
--- Miniterm on /dev/cu.usbmodem1421: 115200,8,N,1 ---
--- Quit: Ctrl+]  |  Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
....
```

Is this intentional? It looks like other settings (`echo`) also don't take effect unless `port` has been set.

ivankravets, 2015-07-07:
Hm... I've just checked the code: https://github.com/platformio/platformio/blob/develop/platformio/commands/serialports.py#L73:L77

PlatformIO will try to find available port automatically. In your case, it can't perform it. Could I ask you to give me an output of this command?
```shell
platformio serialports list
```

efatsi, 2015-07-07:
```
$ platformio serialports list
/dev/cu.Bluetooth-Incoming-Port
-------------------------------
Hardware ID: n/a
Description: n/a

/dev/cu.Bluetooth-Modem
-----------------------
Hardware ID: n/a
Description: n/a

/dev/cu.usbmodem1411
--------------------
Hardware ID: USB VID:PID=2341:43 SNR=74934303030351405141
Description: None
```

It doesn't have any trouble finding the right port if I don't pass in any options, but then of course I haven't set the baud rate I'm looking for and get a bunch of gibberish
```
$ platformio serialports monitor
--- Miniterm on /dev/cu.usbmodem1411: 9600,8,N,1 ---
--- Quit: Ctrl+]  |  Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
?D?x???????????????????????????????????????????????????????????????????????????yy?@6=0????????????????
```

ivankravets, 2015-07-07:
Thanks, I've just reproduced this bug. I'll fix it.

efatsi, 2015-07-07:
Thanks for the quick response! I wish I could contribute but flipping around the code couldn't see where this enforcement was defined.

ivankravets, 2015-07-07:
I've just fixed it :) 

P.S: Try PlatformIO `2.2.1.dev0` http://docs.platformio.org/en/latest/installation.html#development-version

closed 2015-07-07 by ivankravets

efatsi, 2015-07-07:
Thanks again!

commit 76415b93 (2015-07-07): Fixed incorrect arguments handling for "platformio serialports monitor" command // Resolve #248