platformio/platformio-core #249: scons: *** [.pioenvs/engduinov3/firmware.elf] Error 1

emettely, 2015-07-10:
Hi Ivan,

I think there is an error in the code. I cannot compile.

    collect2: error: ld returned 1 exit status
    scons: *** [.pioenvs/engduinov3/firmware.elf] Error 1

I have done `platformio update` and `platformio run`.
Please let me know what is wrong.

ivankravets, 2015-07-10:
Could you provide a code which doesn't work? Or some "sketch". I don't know how to reproduce it.

emettely, 2015-07-10:
Sorry! I will upload more detail

This is the config.ini file:
    [env:engduinov3]
    platform = atmelavr
    framework = arduino
    board = engduinov3

```cpp
/**
* \addtogroup EngduinoLEDs
*
* This is the driver code for LEDs on the Engduino
* These LEDS are not directly connected to pins on the
* AtMega32u4 processor. Instead they are connected through
* LED drivers so should only be accessed through this code.
*
* The Engduino has 16 RGB LEDs on it, each of which
* can be controlled independently. To make the granularity
* of control greater, we implement a software PWM for each
* LED. This allows for 16 levels of brightness in each of
* the RGB channels on each LED with minimal flicker.
* 
* This implementation uses TIMER4 Comparator A to
* implement the PWM. The timer is set to run at ~320Hz. The
* clock is reset on interrupt, so you should be very wary
* about using the other comparators.
*
* @{
*/

/**
* \file 
* 		Engduino LED driver
* \author
* 		Engduino team: support@engduino.org
*/

#include "pins_arduino.h"
#ifdef __BOARD_ENGDUINOV3
	#include <SPI.h>
#endif
#include "EngduinoLEDs.h"

/*---------------------------------------------------------------------------*/
/**
* \brief Constructor
* 
* C++ constructor for this class. Empty.
*/
EngduinoLEDsClass::EngduinoLEDsClass()
{
}

/*---------------------------------------------------------------------------*/
/**
* \brief begin function - must be called before using other functions
*
* The connection to the LEDs is not direct. Instead, it happens through
* three LED drivers - one each for the R, G, and B channels. These are
* connected through a daisy-chained SPI connection. There is a single latch
* line connected to all driver chips, which latches the value set by SPI when
* the line is pulsed high. The output from each driver chip can be switched on
* or off with a separate output enable line to each driver - when these are LOW
* the driver output lines go to the LEDs; when HIGH, they do not.
* 
*/
void EngduinoLEDsClass::begin() 
{
	for (int i = 0; i < 15; i++) {
[... 623 lines cut ...]
	writeLEDs(EngduinoLEDs.RDisp);
#ifdef __BOARD_ENGDUINOV1	
	PORTB |= _BV(PORTB7); 			// Pulse LATCH to switch
	PORTB &= ~_BV(PORTB7);
#elif defined (__BOARD_ENGDUINOV2)
	PORTB |= _BV(PORTB4); 			// Pulse LATCH to switch
	PORTB &= ~_BV(PORTB4);
#elif defined(__BOARD_ENGDUINOV3)	
	LED_LATCH_PORT |= _BV(LED_LATCH_BIT); 			// Pulse LATCH to switch
	LED_LATCH_PORT &= ~_BV(LED_LATCH_BIT);
#endif
	// And we're done with this call of the ISR, so re-enable interrupts
	//
	TIMSK4 |= 0x40;     // And enable interrupts for a compare A match only
}

/*---------------------------------------------------------------------------*/
/*
 * Preinstantiate Objects
 */ 
EngduinoLEDsClass EngduinoLEDs = EngduinoLEDsClass();

/** @} */

```

valeros, 2015-07-13:
Hi @emettely!
Can you provide full error text?
Do you have `setup` and `loop` functions in your project?

emettely, 2015-07-13:
Valeriy,

I have an empty setup function. There are some loops in the EngduinoLEDsClass.
Under is the error code.

-----------------------------------------------------------------------------------------------------------------------------------------------
avr-g++ -o .pioenvs/engduinov3/src/EngduinoLEDs.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=8000000L -DUSB_VID=0x1B4F -DUSB_PID=0x9208 -DUSB_PRODUCT=\"EngduinoV3\" -DARDUINO=10605 -DPLATFORMIO=020201 -I.pioenvs/engduinov3/FrameworkArduino -I.pioenvs/engduinov3/FrameworkArduinoVariant -I.pioenvs/engduinov3/SPI -I.pioenvs/engduinov3/ZZ_EngduinoLEDs src/EngduinoLEDs.cpp
avr-ar rcs .pioenvs/engduinov3/libFrameworkArduinoVariant.a
avr-ranlib .pioenvs/engduinov3/libFrameworkArduinoVariant.a
avr-g++ -o .pioenvs/engduinov3/FrameworkArduino/CDC.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=8000000L -DUSB_VID=0x1B4F -DUSB_PID=0x9208 -DUSB_PRODUCT=\"EngduinoV3\" -DARDUINO=10605 -I.pioenvs/engduinov3/FrameworkArduino -I.pioenvs/engduinov3/FrameworkArduinoVariant .pioenvs/engduinov3/FrameworkArduino/CDC.cpp
avr-g++ -o .pioenvs/engduinov3/FrameworkArduino/HID.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=8000000L -DUSB_VID=0x1B4F -DUSB_PID=0x9208 -DUSB_PRODUCT=\"EngduinoV3\" -DARDUINO=10605 -I.pioenvs/engduinov3/FrameworkArduino -I.pioenvs/engduinov3/FrameworkArduinoVariant .pioenvs/engduinov3/FrameworkArduino/HID.cpp
avr-g++ -o .pioenvs/engduinov3/FrameworkArduino/HardwareSerial.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=8000000L -DUSB_VID=0x1B4F -DUSB_PID=0x9208 -DUSB_PRODUCT=\"EngduinoV3\" -DARDUINO=10605 -I.pioenvs/engduinov3/FrameworkArduino -I.pioenvs/engduinov3/FrameworkArduinoVariant .pioenvs/engduinov3/FrameworkArduino/HardwareSerial.cpp
avr-g++ -o .pioenvs/engduinov3/FrameworkArduino/HardwareSerial0.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=8000000L -DUSB_VID=0x1B4F -DUSB_PID=0x9208 -DUSB_PRODUCT=\"EngduinoV3\" -DARDUINO=10605 -I.pioenvs/engduinov3/FrameworkArduino -I.pioenvs/engduinov3/FrameworkArduinoVariant .pioenvs/engduinov3/FrameworkArduino/HardwareSerial0.cpp
avr-g++ -o .pioenvs/engduinov3/FrameworkArduino/HardwareSerial1.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=8000000L -DUSB_VID=0x1B4F -DUSB_PID=0x9208 -DUSB_PRODUCT=\"EngduinoV3\" -DARDUINO=10605 -I.pioenvs/engduinov3/FrameworkArduino -I.pioenvs/engduinov3/FrameworkArduinoVariant .pioenvs/engduinov3/FrameworkArduino/HardwareSerial1.cpp
avr-g++ -o .pioenvs/engduinov3/FrameworkArduino/HardwareSerial2.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=8000000L -DUSB_VID=0x1B4F -DUSB_PID=0x9208 -DUSB_PRODUCT=\"EngduinoV3\" -DARDUINO=10605 -I.pioenvs/engduinov3/FrameworkArduino -I.pioenvs/engduinov3/FrameworkArduinoVariant .pioenvs/engduinov3/FrameworkArduino/HardwareSerial2.cpp
avr-g++ -o .pioenvs/engduinov3/FrameworkArduino/HardwareSerial3.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=8000000L -DUSB_VID=0x1B4F -DUSB_PID=0x9208 -DUSB_PRODUCT=\"EngduinoV3\" -DARDUINO=10605 -I.pioenvs/engduinov3/FrameworkArduino -I.pioenvs/engduinov3/FrameworkArduinoVariant .pioenvs/engduinov3/FrameworkArduino/HardwareSerial3.cpp
avr-g++ -o .pioenvs/engduinov3/FrameworkArduino/IPAddress.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=8000000L -DUSB_VID=0x1B4F -DUSB_PID=0x9208 -DUSB_PRODUCT=\"EngduinoV3\" -DARDUINO=10605 -I.pioenvs/engduinov3/FrameworkArduino -I.pioenvs/engduinov3/FrameworkArduinoVariant .pioenvs/engduinov3/FrameworkArduino/IPAddress.cpp
avr-g++ -o .pioenvs/engduinov3/FrameworkArduino/Print.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=8000000L -DUSB_VID=0x1B4F -DUSB_PID=0x9208 -DUSB_PRODUCT=\"EngduinoV3\" -DARDUINO=10605 -I.pioenvs/engduinov3/FrameworkArduino -I.pioenvs/engduinov3/FrameworkArduinoVariant .pioenvs/engduinov3/FrameworkArduino/Print.cpp
avr-g++ -o .pioenvs/engduinov3/FrameworkArduino/Stream.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=8000000L -DUSB_VID=0x1B4F -DUSB_PID=0x9208 -DUSB_PRODUCT=\"EngduinoV3\" -DARDUINO=10605 -I.pioenvs/engduinov3/FrameworkArduino -I.pioenvs/engduinov3/FrameworkArduinoVariant .pioenvs/engduinov3/FrameworkArduino/Stream.cpp
avr-g++ -o .pioenvs/engduinov3/FrameworkArduino/Tone.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=8000000L -DUSB_VID=0x1B4F -DUSB_PID=0x9208 -DUSB_PRODUCT=\"EngduinoV3\" -DARDUINO=10605 -I.pioenvs/engduinov3/FrameworkArduino -I.pioenvs/engduinov3/FrameworkArduinoVariant .pioenvs/engduinov3/FrameworkArduino/Tone.cpp
.pioenvs/engduinov3/FrameworkArduino/Tone.cpp:212:12: warning: #warning this may not be correct [-Wcpp]
#warning this may not be correct
^
[... cut ...] ioenvs/engduinov3/libSPI.a .pioenvs/engduinov3/libZZ_EngduinoLEDs.a -Wl,--end-group
.pioenvs/engduinov3/libFrameworkArduino.a(main.o): In function `main':
/Users/user/GitHub/microbit/Eimi/platformIO/.pioenvs/engduinov3/FrameworkArduino/main.cpp:37: undefined reference to `setup'
/Users/user/GitHub/microbit/Eimi/platformIO/.pioenvs/engduinov3/FrameworkArduino/main.cpp:40: undefined reference to `loop'
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/engduinov3/firmware.elf] Error 1

ivankravets, 2015-07-13:
@emettely you don't have defined `setup` and `loop` prototypes, see errors:
```
/Users/user/GitHub/microbit/Eimi/platformIO/.pioenvs/engduinov3/FrameworkArduino/main.cpp:37: undefined reference to `setup'
/Users/user/GitHub/microbit/Eimi/platformIO/.pioenvs/engduinov3/FrameworkArduino/main.cpp:40: undefined reference to `loop'
```

See examples:
* https://github.com/platformio/platformio/blob/develop/examples/wiring-blink/src/main.cpp#L12
* https://github.com/platformio/platformio/blob/develop/examples/wiring-blink/src/main.cpp#L16

emettely, 2015-07-13:
I have added this :
```
void setup(){
	EngduinoLEDs.begin();
}

void loop(){
	EngduinoLEDs.setAll(12, RED, 12);
}
````

and it now works. Thank you so much!

closed 2015-07-14 by ivankravets