platformio/platformio-core #250: Update fails due to Windows Error 145 (dir not empty)

mpd-dude, 2015-07-12:
When calling
<pre>
platformio update
</pre>
under Win7, the following error message and traceback appears:
<pre>
Updating framework-arduinoavr package:
Versions: Current=19, Latest=20          [Out-of-date]
Uninstalling framework-arduinoavr package:      Error: Traceback (most recent call last):
  File "c:\Tools\Python27\lib\site-packages\platformio\__main__.py", line 75, in main
    cli(None, None)
  File "c:\Tools\Python27\lib\site-packages\click\core.py", line 664, in __call__
    return self.main(*args, **kwargs)
  File "c:\Tools\Python27\lib\site-packages\click\core.py", line 644, in main
    rv = self.invoke(ctx)
  File "c:\Tools\Python27\lib\site-packages\click\core.py", line 991, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\Tools\Python27\lib\site-packages\click\core.py", line 837, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\Tools\Python27\lib\site-packages\click\core.py", line 464, in invoke
    return callback(*args, **kwargs)
  File "c:\Tools\Python27\lib\site-packages\platformio\commands\update.py", line 15, in cli
    ctx.invoke(cmd_platforms_update)
  File "c:\Tools\Python27\lib\site-packages\click\core.py", line 464, in invoke
    return callback(*args, **kwargs)
  File "c:\Tools\Python27\lib\site-packages\platformio\commands\platforms.py", line 155, in platforms_update
    p.update()
  File "c:\Tools\Python27\lib\site-packages\platformio\platforms\base.py", line 308, in update
    pm.update(name)
  File "c:\Tools\Python27\lib\site-packages\platformio\pkgmanager.py", line 140, in update
    self.uninstall(name)
  File "c:\Tools\Python27\lib\site-packages\platformio\pkgmanager.py", line 116, in uninstall
    rmtree(join(self._package_dir, name))
  File "c:\Tools\Python27\lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "c:\Tools\Python27\lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "c:\Tools\Python27\lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "c:\Tools\Python27\lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "c:\Tools\Python27\lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "c:\Tools\Python27\lib\shutil.py", line 256, in rmtree
    onerror(os.rmdir, path, sys.exc_info())
  File "c:\Tools\Python27\lib\shutil.py", line 254, in rmtree
    os.rmdir(path)
WindowsError: [Error 145] Das Verzeichnis ist nicht leer: 'C:\\Users\\aaaaa\\.platformio\\packages\\framework-arduinoavr\\libraries\\__cores__\\digispark\\DigisparkTinyPinChange\\examples'
</pre>

Happy to provide more information if needed. 
Thanks and keep up the excellent work!

ivankravets, 2015-07-13:
Which files contains this folder ``C:\\Users\\aaaaa\\.platformio\\packages\\framework-arduinoavr\\libraries\\__cores__\\digispark\\DigisparkTinyPinChange\\examples``?

mpd-dude, 2015-07-14:
Strangly the directory is in fact empty. The behaviour also occurs with other directories, e.g.
<pre>
.platformio\packages\framework-arduinosam\system\CMSIS\Device\ATMEL\sam4s\include
</pre>
the directory is empty, but Error 145 occurs here as well.

After calling platformio update again, the 'include' directory is being deleted, but then the script complains about the parent directory being not empty ('sam4s'). But this is empty, too. Strange.

After calling platformio update a couple of times, it is all resolved and the update process can be finished successfully.

ivankravets, 2015-07-14:
Hm... PlatformIO can't delete folder if it is opened by another application (Explorer, File Manager, etc) in OS. Could I ask you to delete manually whole ``C:\\Users\\aaaaa\\.platformio`` folder and install appropriate platforms again. Then re-test it.

As for "How to reproduce 'update' command?", just open ``C:\\Users\\aaaaa\\.platformio\appstate.json`` and manually reduce "version of package", and run ``platformio update`` command.

For example, see current version:
```json
"framework-arduinosam": {"version": 10, ...}
```

Just, set it to
```
"framework-arduinosam": {"version": 9, ...}
```

mpd-dude, 2015-07-14:
I have reduced the version in appstate.json as you suggested, it did update "framework-arduinosam" but I could not reproduce the behaviour anymore. I guess is that some Windows process had still access to the directory in the earlier case.

I am good for now. Many thanks for looking into this.

ivankravets, 2015-07-14:
Thanks that using PlatformIO! :+1: 

P.S: Don't forget to give us star :star: https://github.com/platformio/platformio/stargazers

closed 2015-07-14 by ivankravets