platformio/platformio-core #252: Cygwin - error message after successful installation

mpd-dude, 2015-07-17:
Hi,
having installed platformio by calling
<pre>
python get-platformio.py
</pre>

under x86_64 <b>Cygwin</b> 2.0.4 (host Win 7), python 2.7.10

<pre>
 ==> Installation process has been successfully FINISHED! <==

 Please RESTART your Terminal Application and run `platformio --help` command.
Traceback (most recent call last):
  File "get-platformio.py", line 145, in <module>
    sys.exit(main())
  File "get-platformio.py", line 138, in main
    "platformio"]))
  File "get-platformio.py", line 69, in exec_python_cmd
    return exec_command([CURINTERPRETER_PATH] + args, shell=IS_WINDOWS).strip()
  File "get-platformio.py", line 64, in exec_command
    raise Exception(err)
Exception: Error: Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/platformio/__main__.py", line 73, in ma                                                                   in
    requests.packages.urllib3.disable_warnings()
AttributeError: 'module' object has no attribute 'packages'
</pre>

Please let me know if you need more information.

ivankravets, 2015-07-21:
Could you try this installer script? https://github.com/platformio/platformio/blob/develop/scripts/get-platformio.py

ivankravets, 2015-07-29:
Please reopen this issue if you still have this problem.

mpd-dude, 2015-07-30:
Thank you. Sorry for the delayed answer. I have tried on the same x86_64 Cygwin 2.0.4 (host Win 7), python 2.7.10 installation with the script you provided:
<pre>==> Installing Python Package Manager ...
Requirement already up-to-date: pip in /usr/lib/python2.7/site-packages
[SUCCESS]

==> Installing PlatformIO and dependencies ...
Requirement already up-to-date: setuptools in /usr/lib/python2.7/site-packages
Collecting platformio
Requirement already up-to-date: requests>=2.4.0 in /usr/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: bottle in /usr/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: pyserial in /usr/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: click>=3.0 in /usr/lib/python2.7/site-packages (from platformio)
Installing collected packages: platformio
Successfully installed platformio-2.2.1
Collecting http://dl.platformio.org/scons.zip
  Downloading http://dl.platformio.org/scons.zip (863kB)
Installing collected packages: scons
  Running setup.py install for scons
Successfully installed scons
[SUCCESS]

 ==> Installation process has been successfully FINISHED! <==

----------------------------------------
Please RESTART your Terminal Application
----------------------------------------

Then run `platformio --help` command.
</pre>

So the installation process looks good. But when I restart Cygwin and type (as being told) 
<pre>platformio --help</pre>
the following error occurs:
<pre>
Error: Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/platformio/__main__.py", line 73, in main
    requests.packages.urllib3.disable_warnings()
AttributeError: 'module' object has no attribute 'packages'
</pre>

Please let me know if you need more information or if I can test more.

mpd-dude, 2015-07-30:
Sorry, but I can not re-open this issue since I didn't close it myself.

ivankravets, 2015-07-31:
Could I ask you to try this build?
```shell
pip install https://github.com/platformio/platformio/archive/feature/issue-252-cygwin.zip
```

mpd-dude, 2015-08-02:
Uninstalled the old version. Then installed the build you provided. Result:
<pre>$ pip install https://github.com/platformio/platformio/archive/feature/issue-252-cygwin.zip
Collecting https://github.com/platformio/platformio/archive/feature/issue-252-cygwin.zip
  Downloading https://github.com/platformio/platformio/archive/feature/issue-252-cygwin.zip
     / 2.5MB 4.1MB/s
Requirement already satisfied (use --upgrade to upgrade): bottle in /usr/lib/python2.7/site-packages (from platformio==2.2.3.dev0)
Requirement already satisfied (use --upgrade to upgrade): click>=3.0 in /usr/lib/python2.7/site-packages (from platformio==2.2.3.dev0)
Requirement already satisfied (use --upgrade to upgrade): pyserial in /usr/lib/python2.7/site-packages (from platformio==2.2.3.dev0)
Requirement already satisfied (use --upgrade to upgrade): requests>=2.4.0 in /usr/lib/python2.7/site-packages (from platformio==2.2.3.dev0)
Building wheels for collected packages: platformio
  Running setup.py bdist_wheel for platformio
  Stored in directory: /home/aaaaa/.cache/pip/wheels/be/ae/4a/2e4f4672f1b57855e8887a436d44397461868e2edc8259a41c
Successfully built platformio
Installing collected packages: platformio
Successfully installed platformio-2.2.3.dev0</pre>
Then typed in<pre>platformio --help</pre>
which results in
<pre>$ platformio
Traceback (most recent call last):
  File "/usr/bin/platformio", line 7, in <module>
    from platformio.__main__ import main
  File "/usr/lib/python2.7/site-packages/platformio/__main__.py", line 10, in <module>
    from requests.packages import urllib3
ImportError: No module named packages
</pre>
Please let me know if you need more information or if I can test more. Thank you.

ivankravets, 2015-08-02:
Hm... Do you see this folder `/usr/lib/python2.7/site-packages/requests/packages`? Does it contain `urllib3` folder?

mpd-dude, 2015-08-02:
Unfortunately there is no "packages" dir. The content of <pre>/usr/lib/python2.7/site-packages/requests/</pre> is:
<pre>__init__.py   auth.py     cookies.py      models.py         structures.py
__init__.pyc  auth.pyc    cookies.pyc     models.pyc        structures.pyc
__init__.pyo  auth.pyo    cookies.pyo     models.pyo        structures.pyo
adapters.py   certs.py    exceptions.py   sessions.py       utils.py
adapters.pyc  certs.pyc   exceptions.pyc  sessions.pyc      utils.pyc
adapters.pyo  certs.pyo   exceptions.pyo  sessions.pyo      utils.pyo
api.py        compat.py   hooks.py        status_codes.py
api.pyc       compat.pyc  hooks.pyc       status_codes.pyc
api.pyo       compat.pyo  hooks.pyo       status_codes.pyo</pre>

ivankravets, 2015-08-02:
:open_mouth: :open_mouth: :open_mouth: 

Give me please an output of:
```
pip list
```

mpd-dude, 2015-08-02:
<pre>bottle (0.12.8)
bsddb3 (6.1.0)
bzr (2.6b2)
bzr-fastimport (0.13.0.final.0)
cffi (0.9.2)
chardet (2.2.1)
click (4.1)
Codeville (0.8.0)
cvs2svn (2.4.0)
dblatex (0.3.4)
fastimport (0.9.2)
git-review (1.24)
logilab-common (0.62.0)
Mako (1.0.1)
MarkupSafe (0.23)
mercurial (3.3)
numpy (1.9.2)
Pillow (2.8.1)
pip (7.1.0)
platformio (2.2.3.dev0)
ply (3.6)
pycparser (2.12)
Pygments (1.6)
pygobject (3.14.0)
Pyrex (0.9.9)
pyserial (2.7)
reportlab (2.7)
requests (2.7.0)
setuptools (18.0.1)
simplejson (3.6.5)
six (1.9.0)
stgit (0.17.1-dirty)
tailor (0.9.35)
urllib3 (1.10.4)
wheel (0.24.0)</pre>
I did a quick search for "urllib3" and this is the result:
<pre>/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3
/lib/python2.7/site-packages/urllib3
/usr/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3
/usr/lib/python2.7/site-packages/urllib3
</pre>

ivankravets, 2015-08-02:
Try
```
pip uninstall requests
pip install requests
```

mpd-dude, 2015-08-03:
Thank you. This helped and now <pre>platformio --help</pre> runs flawlessly :-)

I don't know why the fresh Cygwin installation had this issue with requests. Do you have any idea?

ivankravets, 2015-08-21:
> I don't know why the fresh Cygwin installation had this issue with requests. Do you have any idea?

Sorry, don't have any ideas. I've just added to docs/FAQ/code explanation for end user. Thanks!

closed 2015-08-21 by ivankravets

DefProc, 2015-10-09:
I had a related error that references this issue: 

```
Error: Invalid installation of Python `requests` package`. See < https://github.com/platformio/platformio/issues/252 >
```

thich turned out to be Ubuntu 14.04LTS's install of `python-requests` that was causing a problem. 

Solution: 

```
sudo apt-get remove python-requests
```

ivankravets, 2015-10-09:
Thanks @DefProc. I've just added your tip to solutions list above.

Hkr-Doppelganger, 2017-03-14:
please help
[!] Looks like you dont have python-requests installed. Please install (apt-get install python-requests) and try again.

help with this

Hkr-Doppelganger, 2017-03-14:
root@localhost:~# setoolkit
bash: setoolkit: command not found

MFornander, 2017-05-27:
I'm running into this.

Ubuntu 16LTS
Both Atom or Atom-Beta
Platform-IO installed either with python script or in Atom
Launching the GUI get be stuck in updating core, restart, repeat
Launching in terminal gets me this requirements link
Trying every combo of uninstall/install requirements does nothing
Have been trying for hours (literally)

ivankravets, 2017-05-27:
@MFornander Fixed in https://github.com/platformio/platformio-core/releases/tag/v3.3.1

Please restart Atom

commit 20a0592b (2015-08-21): Fix "AttributeError: 'module' object has no attribute 'packages'" // Resolve #252