platformio/platformio-core #253: Serialports monitor "Ctrl+]" not working

mpd-dude, 2015-07-18:
Hi there,
when starting a serial monitor by 
<pre>
platformio serialports monitor -p COM12
</pre>
under Win7 the "Ctrl+]" is not working. Maybe it is due the German keyboad layout, where "]" is AltGr+9, so the combination of "Ctrl+AltGr+9" is not working.

But "Ctrl+T followed by Ctrl+H" is working just fine. So maybe a good alternative for Quit would be "<b>Ctrl+T followed by Ctrl+Q</b>"? Maybe this is compatible to many different keyboard layouts?

Happy to provide more information or to test. Thanks.

ivankravets, 2015-07-28:
You can redefine hotkey using `--exit-char` option. Please look into
```
platformio serialports monitor --help
Usage: platformio serialports monitor [OPTIONS]

Options:
  -p, --port TEXT       Port, a number or a device name
  -b, --baud INTEGER    Set baud rate, default=9600
  --parity [N|E|O|S|M]  Set parity, default=N
  --rtscts              Enable RTS/CTS flow control, default=Off
  --xonxoff             Enable software flow control, default=Off
  --rts [0|1]           Set initial RTS line state, default=0
  --dtr [0|1]           Set initial DTR line state, default=0
  --echo                Enable local echo, default=Off
  --cr                  Do not send CR+LF, send CR only, default=Off
  --lf                  Do not send CR+LF, send LF only, default=Off
  -d, --debug           Debug received data (escape non-printable chars)
                        #
                        --debug can be given multiple times:
                        # 0: just print
                        what is received
                        # 1: escape non-printable characters,
                        do newlines as unusual
                        # 2: escape non-printable
                        characters, newlines too
                        # 3: hex dump everything
  --exit-char INTEGER   ASCII code of special character that is used to exit
                        the application, default=0x1d
  --menu-char INTEGER   ASCII code of special character that is used to
                        control miniterm (menu), default=0x14
  --quiet               Diagnostics: suppress non-error messages, default=Off
  --help                Show this message and exit.
```

Also, please read more detailed info in [Miniterm](http://pyserial.sourceforge.net/examples.html) documentation.

mpd-dude, 2015-07-30:
Thank you for implementing this. A minor issue:
I have tried with 
<pre>platformio serialports monitor --exit-char 0x11</pre>
which results in
<pre>Error: Invalid value for "--exit-char": 0x11 is not a valid integer</pre>
... although in the help text the <i>hexadecimal</i> representation is given "<b> [...] default=0x1d</b>"

Using the <i>decimal</i> representation 
<pre>platformio serialports monitor --exit-char 17</pre>
everything works fine.

Do you have any plans to store this "exit char" for the user in order to not type it again every time? If not, not a big thing, users can still script around.

Thanks again.

commit 5761cea4 (2015-07-30): Fixed "platformio serialports monitor --help" information with HEX char for hotkeys // Resolve #253
closed 2015-07-30 by ivankravets