platformio/platformio-core #254: No module named pip

mohamed-soubhi, 2015-07-20:
==> Installing Python Package Manager ...
/tmp/tmpfKws_3/pip.zip/pip/_vendor/requests/packages/urllib3/util/ssl_.py:90: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
/tmp/tmpfKws_3/pip.zip/pip/_vendor/requests/packages/urllib3/util/ssl_.py:90: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
Exception:
Traceback (most recent call last):
  File "/tmp/tmpfKws_3/pip.zip/pip/basecommand.py", line 223, in main
    status = self.run(options, args)
  File "/tmp/tmpfKws_3/pip.zip/pip/commands/install.py", line 299, in run
    root=options.root_path,
  File "/tmp/tmpfKws_3/pip.zip/pip/req/req_set.py", line 646, in install
    **kwargs
  File "/tmp/tmpfKws_3/pip.zip/pip/req/req_install.py", line 813, in install
    self.move_wheel_files(self.source_dir, root=root)
  File "/tmp/tmpfKws_3/pip.zip/pip/req/req_install.py", line 1008, in move_wheel_files
    isolated=self.isolated,
  File "/tmp/tmpfKws_3/pip.zip/pip/wheel.py", line 339, in move_wheel_files
    clobber(source, lib_dir, True)
  File "/tmp/tmpfKws_3/pip.zip/pip/wheel.py", line 310, in clobber
    ensure_dir(destdir)
  File "/tmp/tmpfKws_3/pip.zip/pip/utils/__init__.py", line 70, in ensure_dir
    os.makedirs(path)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 13] Permission denied: '/usr/local/lib/python2.7/dist-packages/pip-7.1.0.dist-info'

[FAILURE]

==> Installing PlatformIO and dependencies ...
/usr/bin/python: No module named pip

[FAILURE]
The installation process has been FAILED!

************************************************************************************
this error appeared in my both Ubuntu PC and raspian system

ivankravets, 2015-07-21:
> OSError: [Errno 13] Permission denied: '/usr/local/lib/python2.7/dist-packages/pip-7.1.0.dist-info'

You don't have permission to install python packages. Please look into  http://docs.platformio.org/en/latest/installation.html#super-quick-mac-linux

Have you tried to use ``sudo``?
```shell
sudo python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"

P.S: I will add to installer script more information when permission is denied. Thanks.
```

ivankravets, 2015-07-21:
Resolved in https://github.com/platformio/platformio/commit/b79a79d423715ad246759849f50d35116783cd57

closed 2015-07-21 by ivankravets