platformio/platformio-core #257: miniterm (and by extension, platformio serialports monitor) requires sudo

hoosierEE, 2015-07-28:
If I try

`platformio serialports monitor` 

there is a slight delay (0.5s) and then my prompt returns to normal.  No messages at all.  Prefixing with `sudo` works as expected and gives me a normal, functioning serial monitor.

It appears python's miniterm is to blame (note *permission denied* toward the bottom):

```python
$ python -m serial.tools.miniterm -p /dev/ttyACM0
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/local/lib/python2.7/dist-packages/serial/tools/miniterm.py", line 694, in <module>
    main()
  File "/usr/local/lib/python2.7/dist-packages/serial/tools/miniterm.py", line 649, in main
    repr_mode=options.repr_mode,
  File "/usr/local/lib/python2.7/dist-packages/serial/tools/miniterm.py", line 171, in __init__
    self.serial = serial.serial_for_url(port, baudrate, parity=parity, rtscts=rtscts, xonxoff=xonxoff, timeout=1)
  File "/usr/local/lib/python2.7/dist-packages/serial/__init__.py", line 78, in serial_for_url
    instance.open()
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 289, in open
    self.fd = os.open(self.portstr, os.O_RDWR|os.O_NOCTTY|os.O_NONBLOCK)
OSError: [Errno 13] Permission denied: '/dev/ttyACM0'
```

Unfortunately for me, `platformio serialports monitor` hides these errors, which made debugging a bit harder.  I'm posting this here to let you guys know.  Not sure what the ideal solution would be.  Forwarding minicom's error message would have been helpful for me, but I can see why you might want to make it less verbose for other people (whose minicom presumably doesn't require root privileges).  Maybe add a footnote to the online docs, or even the `--help` option somewhere.

ivankravets, 2015-07-28:
Thanks a lot for report! I'll look into it later. The quick solution to avoid "permission denied" is to add `udev` rules. Please look into documentation (System requirements): http://docs.platformio.org/en/latest/installation.html#system-requirements ("Note" for Linux Users).

hoosierEE, 2015-07-28:
Thanks @ivankravets, I did add the udev rule, but it had no effect.  I should note that I'm running Ubuntu in a chroot, which might prevent the rule from working.

ivankravets, 2015-07-28:
Give please an output of this command:
```
platformio serialports list
```

hoosierEE, 2015-07-28:
Here ya go:

```bash
$ platformio serialports list
/dev/ttyS3
----------
Hardware ID: n/a
Description: ttyS3

/dev/ttyS2
----------
Hardware ID: n/a
Description: ttyS2

/dev/ttyS1
----------
Hardware ID: n/a
Description: ttyS1

/dev/ttyS0
----------
Hardware ID: n/a
Description: ttyS0

/dev/ttyACM0
------------
Hardware ID: USB VID:PID=16c0:048b SNR=440040
Description: ttyACM0
```

ivankravets, 2015-07-28:
Looks like Teensy board? See https://github.com/platformio/platformio/blob/develop/scripts/99-platformio-udev.rules#L50:L54

Could I ask you to add this line to the bottom of `99-platformio-udev.rules`:
```
SUBSYSTEMS=="usb", ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="048b", MODE:="0666"
```

Seems that these lines don't work:
```
# Teensy boards
ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="04[789]?", ENV{ID_MM_DEVICE_IGNORE}="1"
ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="04[789]?", ENV{MTP_NO_PROBE}="1"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="04[789]?", MODE:="0666"
KERNEL=="ttyACM*", ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="04[789]?", MODE:="0666"
```

hoosierEE, 2015-07-28:
On Tue, Jul 28, 2015 at 1:23 PM, Ivan Kravets <notifications@github.com>
wrote:

> SUBSYSTEMS=="usb", ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="048b",
> MODE:="0666"

Gave it a try, no change.  Thanks for your continued effort!
-Alex

ivankravets, 2015-07-28:
Do yo use any log messages from as for "udev" in `dmsg | grep udev`? Have you tried to reboot computer after these changes?

hoosierEE, 2015-07-29:
Before rebooting: no output.

After rebooting: 

```
[    1.779871] systemd-udevd[126]: starting version 208
[    1.804323] udevd[126]: invalid key/value pair in file /lib/udev/rules.d/50-rtc.rules on line 2,starting at character 53 ('r')
```

Oddly, there is no file on my system named `/lib/udev/rules.d/50-rtc.rules`.

ivankravets, 2015-07-29:
Where did you place [99-platformio-udev.rules](https://github.com/platformio/platformio/blob/develop/scripts/99-platformio-udev.rules)?

From that file:
```ini
# This file must be placed at:
# 	/etc/udev/rules.d/99-platformio-udev.rules    (preferred location)
#   	or
# 	/lib/udev/rules.d/99-platformio-udev.rules    (req'd on some broken systems)
#
# To install, type this command in a terminal:
# 	sudo cp 99-platformio-udev.rules /etc/udev/rules.d/99-platformio-udev.rules
#
# Restart "udev" management tool:
#	sudo service udev restart
#		or
# 	sudo udevadm control --reload-rules
#	sudo udevadm trigger
```

Looks like `udevd` didn't load `99-platformio-udev.rules`.

Another solution how to test `99-platformio-udev.rules`, just make "grammar mistake" in this file, after restarting `udevd` should report that file is broken (like with "50-rtc.rules" in your example)

ivankravets, 2015-07-30:
Thanks, I've just fixed it. PlatformIO 2.2.2 will be released today.

closed 2015-07-30 by ivankravets

hoosierEE, 2015-07-30:
Cool, that ought to help if this comes up for anyone else.  Thanks for this.

ivankravets, 2015-07-30:
PlatformIO 2.2.2 is out! Try `platformio upgrade` :smile: 

P.S: Did you resolve issue with udev?

hoosierEE, 2015-07-30:
Will upgrade soon!  I had no luck with udev, but I know I'm an edge case, and this solves my problem well enough for development purposes: 

```bash
alias serial='sudo platformio serialports monitor'
```

ivankravets, 2015-07-30:
You need to ask Chrome OS community how to force udev to use own rules.

Your solutions look good. However, I don't recommend to use "sudo" with PlatformIO. PlatformIO uses `${HOME_DIR}.platformio` to keep state/packages/libs files. And when on this stage it would save some data, it stored it with `root` permission. Then, `platformio` command without `sudo` will have problem.

P.S: This is in theory :smile:

commit 59505cb8 (2015-07-30): Show internal errors from "Miniterm" using "platformio serialports monitor" command // Resolve #257