platformio/platformio-core #258: Unit testing running on the development machine

Oaz, 2015-07-28:
Hello,
Is there any feature supporting unit testing running on the development machine?

Ideally, when I build, part of my source code (the one with the logic, not the interaction with hardware) is compiled and unit tested on the development machine.
This implies no upload to a physical platform and no use of emulator/simulator.

I went through the docs and did not find anything in this area.
An idea would be to create a "unit testing" platform using a native compiler for the development machine. Only the relevant tests and related source code would be compiled and immediately run.
This would add just another target platform for a project.

I might give it a try (I started to read the doc about the platform manifest and build script) but prefer to ask if this makes sense or if it would be better to use some other approach.

Olivier

ivankravets, 2015-07-29:
> Is there any feature supporting unit testing running on the development machine?

Sure! See [tox.ini@testenv](https://github.com/platformio/platformio/blob/develop/tox.ini#L42). How you can run it?
```shell
pip install tox
git clone https://github.com/platformio/platformio.git
cd platformio
tox -e py27
```

Also, please look into Travis CI build: https://travis-ci.org/platformio/platformio/jobs/71406418

ivankravets, 2015-07-29:
I got you, you need cross compiler for desktop platforms. I'm working on it for @SmartAnthill project. However, only Mac OS X will be the first host OS for desktop cross-compiling. Which type of your OS?

ivankravets, 2015-07-31:
@Oaz what do you think about issue #263 ? Will it help you?

Oaz, 2015-08-01:
You are right. The need behind this unit testing thing is desktop cross compiling. Unit testing would be just a library in desktop compilation.
I'm using GNU/Linux so issue #263 is definitely a great plus for my usage of platformio.
If I can do anything to help, just let me know.

ivankravets, 2015-08-01:
> If I can do anything to help, just let me know. If I can do anything to help, just let me know.

Thanks! Could you help us with tool chains which would be used for cross-compiling from Linux 32/64 to Mac/Win/LinuxARM?

If they are native packages from Linux package manager, this is good. We will show additional information which packages should be installed to make cross-compiling. For example, "from host Linux 64 to target Windows 32". 

From the other side will be good to have independent pre-built tool chains (like mingw, or using it) and automatically install it like PlatformIO package.

ivankravets, 2015-09-07:
Looks like this issue was resolved in PlatformIO 2.3 where is support for desktop development platforms. See docs http://docs.platformio.org/en/latest/platforms/index.html#desktop

Please reopen the issue if you still need help.

closed 2015-09-07 by ivankravets