platformio/platformio-core #260: Support for offline/unpublished/private libraries

WillemMali, 2015-07-29:
Hey,

I find myself having to develop closed-source code for work, and I'd like to use Platformio to manage my code.
However, I haven't found a way of installing PlatformIO packages from a custom URL (with authentication) or from an otherwise private source (i.e. a local folder). For this reason I can't really use PlatformIO to the fullest extend in this project.

Is there any way I could still use PlatformIO to manage my C/C++ libraries?

ivankravets, 2015-07-29:
Yes, please install own libraries into [lib_dir](http://docs.platformio.org/en/latest/projectconf.html#lib-dir). PlatformIO will use your libraries even they are not installed via Library Manager.

WillemMali, 2015-07-29:
Oh, thank you!

Perhaps it'd be good if this were documented? I did see the lib_dir option
but didn't know you could put your own libraries there. I do think it's a
rather useful feature.

Regards,
Willem

ivankravets, 2015-07-29:
Thanks, I've just added this info.

commit 3813614d (2015-07-29): Support for off-line/unpublished/private libraries // Resolved #260
closed 2015-07-29 by ivankravets