platformio/platformio-core #261: Where does pio get its frameworks from

thawkins, 2015-07-31:
A question came up about the source of the frameworks used on pio, for the arduino framework we are keen to understand the compatability level vs the arduino ide tool, ie what version of the framework is platformio using? 1.0.6, 1.5.1? 1.6.5 etc. 

I tried reading through the code but i could not find where the frameworks where being downloaded from. 

On an upside,  we are converting the Marlin project to platformio, Marlin is the most widely installed 3d printer control firmware

ivankravets, 2015-07-31:
Which development platform is interested for you? PlatformIO has pre-built Arduino-based framework for different platforms. This information is located in documentation. For example, [packages for AtmelAVR](http://docs.platformio.org/en/latest/platforms/atmelavr.html#packages) development platform, where `framework-arduinoavr` is based on the latest `1.6` branch.

> I tried reading through the code but i could not find where the frameworks where being downloaded from.

Welcome to [PlatformIO Storage](https://sourceforge.net/projects/platformio-storage/).

> On an upside, we are converting the Marlin project to platformio, Marlin is the most widely installed 3d printer control firmware

I'm glad to help you with any questions as for converting. Don't forget that you can integrate your project with Continuos Integration system in "a few clicks". http://docs.platformio.org/en/latest/ci/index.html

See [Travis CI example](http://docs.platformio.org/en/latest/ci/travis.html#examples).

thawkins, 2015-07-31:
1.6 is fine for us...

I notice it is stored on sourceforge, are there plans to move it from there
to something less shaky given its recent history and performamce.

Is the framework version the same for all, for example is
framework-arduinoteensy also based on the same version as
framework-arduinoavr.

ivankravets, 2015-07-31:
> I notice it is stored on sourceforge, are there plans to move it from there
to something less shaky given its recent history and performamce.

Yes, I know about it. However, I don't see here good free CDN alternatives. All PlatformIO packages are mirrored to dl.platformio.org. For example, currently SF doesn't work correctly, it has obsolete packages. I can't update them because SF SSH is offline :cry: 

In this situation, PlatformIO uses self hosted services. See [packages manifest](http://api.platformio.org/packages/manifest).

> Is the framework version the same for all, for example is
framework-arduinoteensy also based on the same version as
framework-arduinoavr.

No, all frameworks are independent. `framework-arduinoavr` is based on https://www.arduino.cc ,where `framework-arduinoteensy` is based on [Teensyduino](https://www.pjrc.com/teensy/td_download.html).

thawkins, 2015-07-31:
Re the teensy frameworks etc, that is a shame, one issue is that we use
teensy for a number of boards we support based on the at90usb1286/1287,
the printrboard in prinrbots is basicaly a teensy20pp system with a bunch
of stepper drivers and sensors attatched, The teensy sourced framework is
somewhat out of date, being based on something they must have snapshoted
around 1.0.6 instead of the 1.6 that the general avr branch is on. Its not
a big deal, its just we have a couple of small known differences that make
life difficult.

Platformio is amazing enough as it is :-), but  i was hoping for it to
right all wrongs, and banish all ill's. At least in the framework version
space :-)

ivankravets, 2015-08-03:
> Platformio is amazing enough as it is :-), but  i was hoping for it to
right all wrongs, and banish all ill's. At least in the framework version
space :-)

They say that "Teensyduino 1.24 supports Arduino versions 1.0.6 and 1.6.1 and 1.6.3 and 1.6.4 and 1.6.5.". However, with PlatformIO you don't need to bother with "Arduino updates". You will be up-to-date with the latest version of Teensy Core (Framework) when use PlatformIO.

Could I close this issue?

thawkins, 2015-08-03:
Yes, i think i have all my answers.

ivankravets, 2015-08-03:
Thanks!

Don't hesitate to contact in future.

closed 2015-08-03 by ivankravets