platformio/platformio-core #262: Raspberry Pi as a platform

thawkins, 2015-07-31:
It would be very cool to support cross compiling to raspberrypi (1 and 2) , with an emulation of the aurduino framework. There is a low level gpio, spi etc twiddling library available  called wiringPi.

http://wiringpi.com

There is another mode the rPi can be used in, which is bare metal mode,  where a firmware image is automaticaly loaded from the sdcard, the pi boot system automaticaly loads a fixed file from a fat formated partition on the sdcard. Which gets your code up and running, there are a number of  c++ frameworks designed for working in this mode, but none that as easy to use as the arduino framework and libraries. As far as i understand only a very small part of the arduino framework is actualy hardware dependant (init, pins, spi ), support for the screen can be either added to the serial lib as a consol, or to the lcd libs as a large graphic lcd device.

ivankravets, 2015-07-31:
> It would be very cool to support cross compiling to raspberrypi (1 and 2) 

See issue #263 

> emulation of the aurduino framework

Sorry, this is out of scope of PlatformIO project. Looks like it should be separate framework/tool which should do it.

ivankravets, 2015-09-07:
@thawkins Have you tried new [native](http://docs.platformio.org/en/latest/platforms/native.html) development platform on RPi? Since PlatformIO 2.3 you can build http://wiringpi.com using `native` development platform and RPi.

ivankravets, 2015-09-21:
Please reopen this issue if you still need help

closed 2015-09-21 by ivankravets

ivankravets, 2015-12-13:
@thawkins Please take a look on issue #372