platformio/platformio-core #263: Cross compiler for desktop platforms

ivankravets, 2015-07-31:
This isn't a secret that PlatformIO initially was intended to be cross compiling tool for multiple development platforms, not just embedded.

According to the latest issues #258 and #262 I understand that this feature for desktop platforms could be interested for the developers. 

# What I propose

## `native` development platform

Create new `native` development platform which will use host OS's `gcc` compiler. If it doesn't exist (like on Windows), PlatformIO will install own pre-built package with `gcc` for Windows. For Mac OS X, Linux - need to show "HELP NOTICE" how `gcc` can be installed using OS's packages manager (`brew`, `apt`, `yum`, etc.)

## Desktop cross-compiling

Create new desktop based platforms:

* `windows32`
* `linux32`
* `linux64`
* `darwin64`
* `linuxarmv6l`
* `linuxarmv7l`

These platforms could be used for cross compiling between desktop platforms. For example, host is Mac OS, then this `platformio.ini` should prepare binaries for targets:
```ini
[env:win]
platform = windows32

[env:mac]
platform = darwin64

[env:lin32]
platform = linux32
```

What do you think about it?

WillemMali, 2015-08-06:
I've been thinking about it myself, seems like a great idea. +1!

ivankravets, 2015-08-09:
@WillemMali which is your host OS?

WillemMali, 2015-08-10:
I'm on Windows (8.1 x64, for entertainment) and Linux (Debian stable x64,  for work). I also regularly work with Raspberry Pi's.

ivankravets, 2015-08-10:
@WillemMali you can try the latest build. It has support for cross-compiling to windows x86 from Mac OS X, Linux 32/64

Automatischa, 2015-09-04:
Biggest security hole once again but must be a relief to non talented coders and starters

ivankravets, 2015-09-04:
Dear talented @Automatischa,

Please explain us about "Biggest security hole once again".

Thanks in advance!

Automatischa, 2015-09-04:
Its only pushing open source once again

Automatischa, 2015-09-04:
There are no rules at all just like everywhere but when i think twice its good like that ;-)

ivankravets, 2015-09-04:
@Automatischa  Thanks, it seems that we speak using different languages. 

Have a nice weekend! :+1:

Automatischa, 2015-09-04:
Rules, not able to delete posts? Bs

closed 2015-09-21 by ivankravets