platformio/platformio-core #266: Raspberry Pi 2 Host System -> Arduino UNO : error: toolchain-atmelavr/bin/avr-g++: No such file or directory

ap0calypse, 2015-08-09:
Hi,

I am trying to connect my Arduino UNO to my Raspberry PI for testing purposes. But something wrong and I can't figure ou what it is. I tried to make the example project from the platformio website and when I run 'platformio run', I get this:

root@xxx:~/new# platformio run    
[Sun Aug  9 18:58:50 2015] Processing uno (platform: atmelavr, board: uno, framework: arduino)
-----------------------------------------------------------------------------------------------------------------------
avr-g++ -o .pioenvs/uno/src/main.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10605 -DPLATFORMIO=020202 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant src/main.cpp
sh: /root/.platformio/packages/toolchain-atmelavr/bin/avr-g++: No such file or directory
scons: *** [.pioenvs/uno/src/main.o] Error 127
============================================= [ ERROR ] Took 5.09 seconds =============================================

I looked up if the file " /root/.platformio/packages/toolchain-atmelavr/bin/avr-g++" exists, and it does. So far everything _should_ work, I don't know what's the problem.

Could you help? Thanks!

ivankravets, 2015-08-09:
Hi @ap0calypse ,

Why do you run @PlatformIO with `root` permissions? @PlatformIO doesn't need in super-user permissions. 

1. Remove totally `/root/.platformio` folder.
2. Try the example with general user without `sudo`.

ap0calypse, 2015-08-09:
Hi @ivankravets , thanks for the quick reply :)

I tried the example again, but the result is still the same. Here is my output:

>testuser@xxx:~/test$ platformio run
>[Sun Aug  9 19:15:35 2015] Processing uno (platform: atmelavr, board: uno, framework: arduino)
>-----------------------------------------------------------------------------------------------------------------------
>The platform 'atmelavr' has not been installed yet. Would you like to install it now? [y/N]: y
>Installing toolchain-atmelavr package:
>Downloading  [####################################]  100%          
>Unpacking  [####################################]  100%          
>Installing tool-avrdude package:
>Downloading  [####################################]  100%
>Unpacking  [####################################]  100%
>Installing framework-arduinoavr package:
>Downloading  [####################################]  100%          
>Unpacking  [####################################]  100%          
>Installing tool-micronucleus package:
>Downloading  [####################################]  100%
>Unpacking  [####################################]  100%
>The platform 'atmelavr' has been successfully installed!
>avr-g++ -o .pioenvs/uno/src/main.o -c -fno-exceptions -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10605 -DPLATFORMIO=020202 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant src/main.cpp
>sh: /home/ap0calypse/.platformio/packages/toolchain-atmelavr/bin/avr-g++: No such file or directory
>scons: *** [.pioenvs/uno/src/main.o] Error 127
>============================================ [ ERROR ] Took 143.29 seconds ============================================
testuser@xxx:~/test$

ivankravets, 2015-08-09:
@valeros, please help @ap0calypse

ivankravets, 2015-08-10:
@ap0calypse I have this RPi 2. Which is your distributive?

ap0calypse, 2015-08-10:
@ivankravets I have SlackwareARM-current running on it

ivankravets, 2015-08-14:
Could I ask you to install this:
```
sudo apt-get install libmpc-dev
```

ivankravets, 2015-08-14:
I've just installed latest Raspbian distributive on RPi 2 - @PlatformIO works like a charm!

ap0calypse, 2015-08-16:
Hi, thanks for responding.

1. I have no apt-get on my system, but I have libmpc 1.0.3 installed.
2. as already mentioned, I am running SlackwareARM, not Raspbian.

Thanks for your time anyway :)

ivankravets, 2015-08-27:
Could you provide access to this SlackwareARM? Mail me@ikravets.com.
Thanks.

ivankravets, 2015-09-07:
@ap0calypse  Please reopen this issue if you still need help.

closed 2015-09-07 by ivankravets