platformio/platformio-core #267: Compile Arduino with C++11

efatsi, 2015-08-10:
Is it possible with the current platformio to compile Arduino projects with support for C++11? I believe it should be as easy as adding a `-std=gnu++11` flag to the cpp compile step.

ivankravets, 2015-08-10:
Have you tried [build_flags](http://docs.platformio.org/en/latest/projectconf.html#projectconf-build-flags) option?

efatsi, 2015-08-10:
Ah sorry, should have checked the docs more closely. Thanks.

closed 2015-08-10 by efatsi

ivankravets, 2015-08-10:
Sorry for the off-top: Could I ask you to look into https://github.com/vigetlabs/platomformio/pull/2 and release new version of Atom package with all changes?

Thanks in advance!

efatsi, 2015-08-10:
merged and pushed

ivankravets, 2015-08-10:
Could you add description for the repo and URL to Atom's site? See my fork: https://github.com/platformio/platomformio

Like this (top, under number of commits):
```
Integration Atom IDE with PlatformIO https://atom.io/packages/platomformio  — Edit
```