platformio/platformio-core #268: can't upload to TeensyLC

altafferw, 2015-08-12:
unable to upload to TeensyLC:
-------------
Found HalfKay Bootloader
error writing to Teensy
scons: *** [upload] Error 1
-------------

Simple fix, pull latest [PJRC command line loader] code from:  https://github.com/PaulStoffregen/teensy_loader_cli

ivankravets, 2015-08-12:
Which is type of your host OS?

altafferw, 2015-08-12:
Greetings,

> Which is type of your host OS?

- Debian 8.0 Linux, 64 bit, quad core
- Python 2.7.10

uname -a => Linux 4.1.0-1-amd64 #1 SMP Debian 4.1.3-1 (2015-08-03)
x86_64 GNU/Linux

Also have Arduino 1.6.5 w/Teensyduino 1.24 installed.

Later,
	William Altaffer
	altafferw@gmail.com

ivankravets, 2015-08-13:
Please update PlatformIO packages:
```shell
platformio update
```

Has it helped you?

altafferw, 2015-08-13:
Greetings,

> Please update PlatformIO packages:
> platformio update
> 
> Has it helped you?

That fixed it.  Looks like you (or I?) can close "issue #268".

Thanks,
	William Altaffer
	altafferw@gmail.com

ivankravets, 2015-08-13:
Thanks @valeros for updates!
Thanks @altafferw for report! :beers:

closed 2015-08-13 by ivankravets