platformio/platformio-core #269: Install through proxy

frankvdh, 2015-08-19:
Reporting as requested during the install. How do I specify my proxy server name, port, username & password to the installer?

C:\Users\frankv\Downloads>C:\Python27\python.exe get-platformio.py

==> Fixing Windows %PATH% Environment ...
[SUCCESS]

==> Installing Python Package Manager ...
<urlopen error Tunnel connection failed: 407 Proxy Access Denied>
[FAILURE]
The installation process has been FAILED!
Please report about this problem here
< https://github.com/platformio/platformio/issues >

C:\Users\frankv\Downloads>

frankvdh, 2015-08-19:
Answering my own question...
The following almost works...

1. Download https://bootstrap.pypa.io/get-pip.py
2. Run C:\Python27\python.exe get-pip.py --no-wheel --no-setuptools
    You may get "Requirement already satisfied"... that's OK
3. Run pip install setuptools --proxy [user]:[password]@[proxy.server]:[port]
    You may get "Requirement already satisfied"... that's OK

    Except it fails at this step with proxy errors, even though I'm specifying the proxy details :(
4. Run pip install platformio --proxy [user]:[password]@[proxy.server]:[port]

ivankravets, 2015-08-19:
Try this http://stackoverflow.com/questions/11726881/how-to-set-an-http-proxy-in-python-2-7

ivankravets, 2015-08-21:
Has it helped you? If yes, please close this issue.

frankvdh, 2015-08-21:
I've worked through several of those ideas, but haven't yet got it working.

ivankravets, 2015-08-21:
As I've got you, you have successfully installed `pip` and `setuptools`?

What is interesting, that you installed `setuptools` via `pip` without problems. However, you can't install `platformio` package. Am I right?

frankvdh, 2015-08-22:
I think setuptools was already installed with Python

ivankravets, 2015-08-22:
Give me please an output of this command 
```
pip list
```

frankvdh, 2015-08-24:
Below is pip list command and also attempt to install platformio with
verbose output.

Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

C:\Users\frankv>pip list
pip (7.0.1)
setuptools (16.0)

C:\Users\frankv>pip install -v platformio --proxy frankv:password@pn-proxy10
:8080
Starting new HTTPS connection (1): pypi.python.org
There was an error checking the latest version of pip
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\pip\utils\outdated.py", line 122, in
pip_version_check
    headers={"Accept": "application/json"},
  File "C:\Python27\lib\site-packages\pip\_vendor\requests\sessions.py",
line 477, in get
    return self.request('GET', url, **kwargs)
  File "C:\Python27\lib\site-packages\pip\download.py", line 373, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "C:\Python27\lib\site-packages\pip\_vendor\requests\sessions.py",
line 465, in request
    resp = self.send(prep, **send_kwargs)
  File "C:\Python27\lib\site-packages\pip\_vendor\requests\sessions.py",
line 573, in send
    r = adapter.send(request, **kwargs)
  File "C:\Python27\lib\site-packages\pip\_vendor\cachecontrol\adapter.py",
line 46, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "C:\Python27\lib\site-packages\pip\_vendor\requests\adapters.py",
line 424, in send
    raise ConnectionError(e, request=request)
ConnectionError: HTTPSConnectionPool(host='pypi.python.org', port=443): Max
retries exceeded with url: /pypi/pip/json (Caused by Pro
xyError('Cannot connect to proxy.', error(10061, 'No connection could be
made because the target machine actively refused it')))
Collecting platformio
  Getting page https://pypi.python.org/simple/platformio/
  Starting new HTTPS connection (1): pypi.python.org
  Incremented Retry for (url='/simple/platformio/'): Retry(total=4,
connect=None, read=None, redirect=None)
  Retrying (Retry(total=4, connect=None, read=None, redirect=None)) after
connection broken by 'ProxyError('Cannot connect to proxy.
', error(10061, 'No connection could be made because the target machine
actively refused it'))': /simple/platformio/
  Starting new HTTPS connection (2): pypi.python.org
  Incremented Retry for (url='/simple/platformio/'): Retry(total=3,
connect=None, read=None, redirect=None)
  Retrying (Retry(total=3, connect=None, read=None, redirect=None)) after
connection broken by 'ProxyError('Cannot connect to proxy.
', error(10061, 'No connection could be made because the target machine
actively refused it'))': /simple/platformio/
  Starting new HTTPS connection (3): pypi.python.org
  Incremented Retry for (url='/simple/platformio/'): Retry(total=2,
connect=None, read=None, redirect=None)
  Retrying (Retry(total=2, connect=None, read=None, redirect=None)) after
connection broken by 'ProxyError('Cannot connect to proxy.
', error(10061, 'No connection could be made because the target machine
actively refused it'))': /simple/platformio/
  Starting new HTTPS connection (4): pypi.python.org
  Incremented Retry for (url='/simple/platformio/'): Retry(total=1,
connect=None, read=None, redirect=None)
  Retrying (Retry(total=1, connect=None, read=None, redirect=None)) after
connection broken by 'ProxyError('Cannot connect to proxy.
', error(10061, 'No connection could be made because the target machine
actively refused it'))': /simple/platformio/
  Starting new HTTPS connection (5): pypi.python.org
[... 62 lines cut ...]
Exception information:
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\pip\basecommand.py", line 223, in main
    status = self.run(options, args)
  File "C:\Python27\lib\site-packages\pip\commands\install.py", line 280,
in run
    requirement_set.prepare_files(finder)
  File "C:\Python27\lib\site-packages\pip\req\req_set.py", line 317, in
prepare_files
    functools.partial(self._prepare_file, finder))
  File "C:\Python27\lib\site-packages\pip\req\req_set.py", line 304, in
_walk_req_to_install
    more_reqs = handler(req_to_install)
  File "C:\Python27\lib\site-packages\pip\req\req_set.py", line 439, in
_prepare_file
    req_to_install.populate_link(finder, self.upgrade)
  File "C:\Python27\lib\site-packages\pip\req\req_install.py", line 244, in
populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "C:\Python27\lib\site-packages\pip\index.py", line 563, in
find_requirement
    'No matching distribution found for %s' % req
DistributionNotFound: No matching distribution found for platformio

C:\Users\frankv>

ivankravets, 2015-08-25:
Sorry, looks like you have problems with your proxy provider. I've just tried [free proxy](http://www.freeproxylists.net) and see my result:
```
➜ pip install --proxy 5.45.122.98:3128 platformio
Collecting platformio
Collecting requests>=2.4.0 (from platformio)
  Using cached requests-2.7.0-py2.py3-none-any.whl
Collecting bottle (from platformio)
Collecting pyserial (from platformio)
Collecting click>=3.0 (from platformio)
  Using cached click-5.1-py2.py3-none-any.whl
Installing collected packages: requests, bottle, pyserial, click, platformio
Successfully installed bottle-0.12.8 click-5.1 platformio-2.2.2 pyserial-2.7 requests-2.7.0
```

Let's resolve it step by step. First need to install latest ``pip``:
1. Download directly `pip` from https://pypi.python.org/packages/source/p/pip/pip-7.1.2.tar.gz
2. Run this:
```shell
pip install pip-7.1.2.tar.gz
```
3. Now try pip with your proxy server:
```shell
pip install -v --proxy frankv:password@pn-proxy10
:8080 platformio 
```

P.S: Do you have special symbols in password? Maybe need to quote them? Like, `` --proxy "frankv:password@pn-proxy10"``?

ivankravets, 2015-08-31:
@frankvdh please reopen this issue if you still has problem with proxy & pip.

closed 2015-08-31 by ivankravets