platformio/platformio-core #271: No custom include path in compiler command line

aunoor, 2015-08-20:
If add to src_build any -I directive containing relative path, then platformio parser just throw it away.
Example:
`src_build = -I/home/user/include` - work, if dir exist
`src_build = -I.` - don't work

TD22057, 2015-09-04:
This is also an issue on windows.  I build a default project with SublimeText which creates a src and lib directory but the lib directory is not put in the path.  So I tried adding it to the config file and can't figure out any way to do it on Windows 7, with platformio 2.2.2.  Here are the various things I tried and what I see on the compiler line:

Input: src_build_flags = -Ic:/Users/foo/project/lib
Compiler: -Ilib

Input: src_build_flags = -I/Users/foo/project/lib
Compiler: -Ilib

Input: src_build_flags = -I../lib
Compiler: -Ilib

Is there any work around?  This is pretty crippling at this point.

ivankravets, 2015-09-05:
@TD22057 PlatformIO automatically handles libraries from `lib` directory. Each lib should be placed in own directory. See docs http://docs.platformio.org/en/latest/userguide/cmd_init.html#description

As for this issue - I'll work on it. Sorry for the troubles :cry:

TD22057, 2015-09-05:
Thanks.  Off topic: I was hoping to not have a flat library (lib/LIB_NAME/SOURCE) structure and use platformio.  There are way too many libraries with the same names so I was hoping to have lib/DIR/LIB_NAME/SOURCE and then say "#include "DIR/LIB_NAME/File.h" in my code.  Of course I'm coming from a regular C++ dev background so perhaps that's the wrong way to think about Arduino...

ivankravets, 2015-09-05:
@TD22057 PlatformIO looks for source code of libraries in `lib/DIRs`. Try this:
```c
#include <LIB_NAME/File.h>
```

* don't need additional flags
* PlatformIO will automatically find dependencies from your source files and built static libs. You should see it in build result.

Please report me if you still have problems.

closed 2015-11-27 by ivankravets

ivankravets, 2015-11-27:
Dear @aunoor, @TD22057,

Please re-test the latest PlatformIO 2.3.6.dev3 http://docs.platformio.org/en/latest/installation.html#development-version

commit 221b7ed1 (2015-11-27): Fix relative include path for preprocessor using "build_flags" // Resolve #271