platformio/platformio-core #272: IDE integration fails for Eclipse OSX Luna SR2 or MARS - missing builder

gslender, 2015-08-20:
IDE integration fails for Eclipse OSX Luna SR2 or MARS - missing builder with a path not found.

Error is [Program "platformio" not found in PATH]

If I copy full/correct path it works..../usr/local/bin/platformio

Not sure why

gslender, 2015-08-20:
Didn't get any better after that...

08:09:11 **** Build of configuration Default for project art.test ****
/usr/local/bin/platformio --force run 
[Fri Aug 21 08:09:11 2015] Processing uno (platform: atmelavr, targets: upload, framework: arduino, board: uno)
--------------------------------------------------------------------------------
Error: The PlatformIO and `scons` aren't installed properly. Please use official installation manual: http://docs.platformio.org/en/latest/installation.html
Command canceled

08:09:14 Build Cancelled (took 2s.475ms)

gslender, 2015-08-20:
I've had both Windows and now Mac OSX not work in terms of integration - talk about bad luck!

gslender, 2015-08-21:
I think this issue is due to the fact that Mac OS X does not include /usr/local/bin in its default PATH. Therefore, assuming you're using the default bash shell, you need to execute the command (or include in /etc/profile) export PATH=/usr/local/bin:$PATH

gslender, 2015-08-21:
Fixed the problem by adding setenv PATH /usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin to the file /etc/launchd.conf and rebooted.... Eclipse now can find platformio

ivankravets, 2015-08-21:
Thanks! I've just added info/note to each IDE into docs http://docs.platformio.org/en/latest/ide/eclipse.html#integration

ivankravets, 2015-08-21:
... and explanation to FAQ http://docs.platformio.org/en/latest/faq.html#faq-troubleshooting-pionotfoundinpath

closed 2015-08-21 by ivankravets

gslender, 2015-08-21:
In researching this issue, the solution is different for each release of OS X, so I would recommend that you look at some other way of having Eclipse or OS X find the platformio command. It would seem that /usr/local/bin isn't a place where GUI apps commonly have in their path... a problem that starts in early version of OS X and continues into the latest. The solution I use won't work on OS X 10.10 for example.

Can you put a sym link into /usr/bin or similar upon installation so that it is found in the GUI app on OS X ????

gslender, 2015-08-22:
I had this problem again when installing on a Yosemite (10.10) OS X workstation and fixed it in a way that I think you could probably add to your install script.

```
ln -s /usr/local/bin/platformio platformio
ln -s /usr/local/bin/scons scons
```

ivankravets, 2015-08-22:
Thanks, docs updated.

commit bf15ba3a (2015-08-21): Add answer to FAQ when Program “platformio” not found in PATH // Resolve #272