platformio/platformio-core #273: Don't understand how /lib in the project folder works???

gslender, 2015-08-21:
When I've created a new project, there is a /src and /lib folder, but cannot follow how you are supposed to the use the /lib folder.

In Eclipse, if you place source files in the /lib folder, they are not configured to be used. Also, when you use the Library Manager, it adds the files to the ~/.platformio folder but that doesn't use the /lib folder either... so what gives???

ivankravets, 2015-08-21:
> In Eclipse, if you place source files in the /lib folder, they are not configured to be used.

Please read info in documentation http://docs.platformio.org/en/latest/projectconf.html#lib-dir

> Also, when you use the Library Manager, it adds the files to the ~/.platformio folder but that doesn't use the /lib folder either... so what gives???

PlatformIO Library Manager uses own external folder (`~/.platformio/lib`) which is common between all your projects.

gslender, 2015-08-21:
I still don't follow as when I use the PlatformIO Library Manager it doesn't put anything in the /lib folder at all. Just to be clear, this is the /lib folder that is created next to the /src folder. If I put any .h / .cpp files in that /lib folder then neither PlatformIO or Eclipse is aware of them.

Yes, when I use PlatformIO Library Manager it creates the libraries under ~/.platformio folder but they are not added to Eclipse so you get lots of CODAN errors and it doesn't update the includes used.

Overall, I don't see how using PlatformIO Library Manager helps when using Eclipse... as such, I'm not using it as it doesn't help, it only hinders and was hoping you had better examples or information on this.

ivankravets, 2015-08-21:
Please read `note` in http://docs.platformio.org/en/latest/projectconf.html#lib-dir

> You can put here your own/private libraries. The source code of each library should be placed in separate directory. For example, `%lib_dir%/private_lib/[here are source files]`.

> Yes, when I use PlatformIO Library Manager it creates the libraries under ~/.platformio folder but they are not added to Eclipse so you get lots of CODAN errors and it doesn't update the includes used.

Give me please an output of this command:
```
platformio lib list
```

gslender, 2015-08-21:
I still don't follow why my project .\lib folder is empty - what is the point of that...?? So the only benefit is that I create another folder within it, private_lib/ and then put in source files??? Why the extra folder for no reason. It is empty and all the source files from 3rd party libraries that PlatformIO Library Manager downloads are created in the libraries under ~/.platformio folder ??

```
Grants-MacBook-Pro:megasquirt-lcd-duino grant$ platformio lib list
[ ID  ] Name             Compatibility         "Authors": Description
------------------------------------------------------------------------------------------------------------------------
[ 13  ] Adafruit-GFX     arduino, atmelavr     "Adafruit Industries": A core graphics library for all our displays, providing a common set of graphics primitives (points, lines, circles, etc.)
[ 32  ] Adafruit-TFT-LCD arduino, atmelavr     "Adafruit Industries": Arduino library for 8-bit TFT LCDs such as ILI9325, ILI9328, etc
Grants-MacBook-Pro:megasquirt-lcd-duino grant$ 
```

Despite all that, there is no update the any Eclipse project files... so how does the Eclipse project know about all the /lib folder... either via PlatformIO Library Manager or the /lib/private_lib folder?

gslender, 2015-08-21:
Here is my projects' folders... I'm not sure if we're talking about the same /lib folder... I mean the one in this directory listing below... NOT the lib folder at ~/.platformio

```
Grants-MacBook-Pro:megasquirt-lcd-duino grant$ ls -l
total 24
-rw-r--r--   1 grant  staff  1081 20 Aug 21:08 LICENSE
-rw-r--r--   1 grant  staff    23 20 Aug 21:08 README.md
drwxr-xr-x   2 grant  staff    68 20 Aug 21:09 lib   <<<<<< what is this folder for??? it is always empty?
-rw-r--r--   1 grant  staff   507 20 Aug 21:09 platformio.ini
drwxr-xr-x  28 grant  staff   952 21 Aug 21:06 src
Grants-MacBook-Pro:megasquirt-lcd-duino grant$ 
```

ivankravets, 2015-08-21:
> I still don't follow why my project .\lib folder is empty - what is the point of that...?? So the only benefit is that I create another folder within it, private_lib/ and then put in source files??? Why the extra folder for no reason.

See in documentation http://docs.platformio.org/en/latest/userguide/cmd_init.html
This folder can be used for private, 3-rd party and etc libraries. The libraries from this folder are visible just for this project.

> It is empty and all the source files from 3rd party libraries that PlatformIO Library Manager downloads are created in the libraries under ~/.platformio folder ??

Yes, because all libraries from PlatformIO Library Manager are visible for all PlatformIO-based project in your system, If you need specific/private/isolated library per project, please use `lib` folder from project.

gslender, 2015-08-21:
ok, I finally seem to have got what is going on... and Eclipse is working. Initially things were getting mixed up with old source files etc.... thanks for persisting with me on this.

I think the docs are little vague on all this and could use a strong set of examples or tutorial on how you'd use it with Eclipse etc.... I got there eventually, but it wasn't clear what was going wrong etc.

G

gslender, 2015-08-21:
Sorry - thought this was sorted but it isn't.... I've dropped a few customer .H .CPP files into the projects /lib folder and Eclipse is fine with finding them. PlatformIO cannot find them, so I can only assume that whilst it creates the /lib folder on project creation, it won't use it as a search path for source files. That seems odd and I don't follow why that would be? Why create a folder that isn't used?

gslender, 2015-08-21:
ok, figured it out... I needed to put my lib files into /lib/private_lib and then platformio found them!!

That isn't clear and you should probably consider having platformio init create that folder too... what's the point of the /lib folder in the project if you can't drop in the files unless they are in the private_lib folder??? 

I worked it out by trial and error and none of the documentation says to do that step. It talks about adding private_lib to the %lib_dir% folder, but that's not the projects lib folder. Thats two different locations.

ivankravets, 2015-08-21:
PlatformIO uses own source code builder. Eclipse is used just for syntax highlighting, code navigation and autocomplete. 

When you build project via Eclipse using PlatformIO, Eclipse delegates all control to PlatformIO. 

Don't afraid if Eclipse highlights that something is wrong. You should relay upon PlatformIO console output.

ivankravets, 2015-08-21:
Please don't close this issue. I'll add this information to documentation. Thanks! :beer;

gslender, 2015-08-21:
Yeah, well I personally think the issue isn't just documentation - I still don't get why the /lib folder is even created alongside the /src folder if you cannot use it without putting things inside the private_lib folder?

All that doesn't make sense and I'm not sure why you wouldn't consider having platformio init create that folder too... how else can you use the lib folder? Does everything need to be in a folder before it is included?

ivankravets, 2015-08-22:
The `readme.txt` file will be created in `lib` directory with this content:
```
Documentation: http://docs.platformio.org/en/latest/userguide/cmd_init.html

This directory is intended for the project specific (private) libraries.
PlatformIO will compile them to static libraries and link to executable file.

The source code of each library should be placed in separate directory.
For example, "lib/private_lib/[here are source files]".
```

ivankravets, 2015-08-22:
Also, added `note` to http://docs.platformio.org/en/latest/userguide/cmd_init.html

SimonRydell, 2016-06-16:
I'm sorry for this incredibly newbie question, but I haven't gotten any of the libraries I installed via PlatformIO to work. whenever I use for example "#include <ESP8266.h> ", I only get the following error:

>scons: *** [.pioenvs/nodemcuv2/src/main.o] Error 1

Do I have to do anything more than just "pio lib install 127" in the commandline to be able to use the libraries? I include the result from writing "pio lib list" below;

> [ 115 ] DHT22            arduino, atmelavr     "Ben Adams, Craig Ringer": Digital-output relative humidity & temperature
 sensor/module DHT22
[ 127 ] ESP8266          arduino, atmelavr     "ITEAD Studio": ESP8266 offers a complete and self-contained Wi-Fi networ
king solution, allowing it to either host the application or to offload all Wi-Fi networking functions from another appl
ication processor
[ 549 ] WebSockets       arduino, atmelavr, atmelsam, timsp430, titiva, teensy, freescalekinetis, ststm32, nordicnrf51,
nxplpc, espressif, siliconlabsefm32, linux_arm, native, intel_arc32 "Markus Sattler": WebSocket Server and Client for Ar
duino based on RFC6455
[1091 ] esp8266_mdns     arduino, espressif    "dunk": mDNS queries and responses on esp8266. Or to describe it another
way: An mDNS Client or Bonjour Client library for the esp8266.
[1101 ] ESP8266wifi      arduino, atmelavr     "Jonas Ekstrand": ESP8266 Arduino library with built in reconnect functio
nality

ivankravets, 2016-06-16:
@SimonRydell try to remove spaces in `< ESP8266.h >` and please share full log output

SimonRydell, 2016-06-16:
**Configuration**

Operating system: Mac OS 10.11.4

PlatformIO Version 2.10.1

IDE Atom with PlatformIO Version 1.2.2

**Description of problem**

I'm sorry for this incredibly newbie question, but I haven't gotten any of the libraries I installed via PlatformIO to work. whenever I use for example "#include < ESP8266.h >" , I only get the following error:

>scons: *** [.pioenvs/nodemcuv2/src/main.o] Error 1

Do I have to do anything more than just "pio lib install 127" in the commandline to be able to use the libraries? I include the result from writing "pio lib list" below;

> [ 115 ] DHT22            arduino, atmelavr     "Ben Adams, Craig Ringer": Digital-output relative humidity & temperature sensor/module DHT22
[ 127 ] ESP8266          arduino, atmelavr     "ITEAD Studio": ESP8266 offers a complete and self-contained Wi-Fi networking solution, allowing it to either host the application or to offload all Wi-Fi networking functions from another application processor

SimonRydell, 2016-06-16:
[Thu Jun 16 23:05:38 2016] Processing nodemcuv2 (platform: espressif, board: nodemcuv2, framework: arduino)

--------------------------------------------------------------------------------
xtensa-lx106-elf-g++ -o .pioenvs/nodemcuv2/src/main.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-sect
ion-literals -falign-functions=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPL
ATFORMIO=021001 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDUINO=20200 -U__STRICT_ANSI__ -I/Users/Sim
onRydell/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I/Users/SimonRydell/.platformio/packages/fram
ework-arduinoespressif/tools/sdk/lwip/include -I.pioenvs/nodemcuv2/FrameworkArduino -I.pioenvs/nodemcuv2/FrameworkArduino
Variant -I.pioenvs/nodemcuv2/SoftwareSerial -I.pioenvs/nodemcuv2/ESP8266_ID127 -Isrc src/main.cpp
<command-line>:0:9: error: expected identifier before numeric constant
.pioenvs/nodemcuv2/ESP8266_ID127/ESP8266.h:38:7: note: in expansion of macro 'ESP8266'
class ESP8266 {
^
<command-line>:0:9: error: expected unqualified-id before numeric constant
.pioenvs/nodemcuv2/ESP8266_ID127/ESP8266.h:38:7: note: in expansion of macro 'ESP8266'
class ESP8266 {
^
<command-line>:0:9: error: expected unqualified-id before numeric constant
src/main.cpp:25:1: note: in expansion of macro 'ESP8266'
ESP8266 wifi(Serial1);

^
src/main.cpp: In function 'void setup()':
src/main.cpp:33:20: error: 'wifi' was not declared in this scope
Serial.println(wifi.getVersion().c_str());
^
xtensa-lx106-elf-gcc -o .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_wiring.o -c -std=gnu99 -Wpointer-arith -Wno-impl
icit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functio
ns=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=021001 -DESP8266 -D
ARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDUINO=20200 -U__STRICT_ANSI__ -I/Users/SimonRydell/.platformio/package
s/framework-arduinoespressif/tools/sdk/include -I/Users/SimonRydell/.platformio/packages/framework-arduinoespressif/tools
/sdk/lwip/include -I.pioenvs/nodemcuv2/FrameworkArduino -I.pioenvs/nodemcuv2/FrameworkArduinoVariant .pioenvs/nodemcuv2/F
rameworkArduino/core_esp8266_wiring.c
xtensa-lx106-elf-gcc -o .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_wiring_analog.o -c -std=gnu99 -Wpointer-arith -W
no-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-
functions=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=021001 -DESP
8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDUINO=20200 -U__STRICT_ANSI__ -I/Users/SimonRydell/.platformio/
packages/framework-arduinoespressif/tools/sdk/include -I/Users/SimonRydell/.platformio/packages/framework-arduinoespressi
f/tools/sdk/lwip/include -I.pioenvs/nodemcuv2/FrameworkArduino -I.pioenvs/nodemcuv2/FrameworkArduinoVariant .pioenvs/node
mcuv2/FrameworkArduino/core_esp8266_wiring_analog.c
scons: *** [.pioenvs/nodemcuv2/src/main.o] Error 1
xtensa-lx106-elf-gcc -o .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_wiring_digital.o -c -std=gnu99 -Wpointer-arith -
Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign
-functions=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=021001 -DES
P8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDUINO=20200 -U__STRICT_ANSI__ -I/Users/SimonRydell/.platformio
/packages/framework-arduinoespressif/tools/sdk/include -I/Users/SimonRydell/.platformio/packages/framework-arduinoespress
if/tools/sdk/lwip/include -I.pioenvs/nodemcuv2/FrameworkArduino -I.pioenvs/nodemcuv2/FrameworkArduinoVariant .pioenvs/nod
emcuv2/FrameworkArduino/core_esp8266_wiring_digital.c
========================= [ ERROR ] Took 1.14 seconds =========================
 
PlatformIO: BuildFile 3Project 88 Issuessrc/main.cpp22:11
LFUTF-8C++

ivankravets, 2016-06-16:
Do you have `#include <Arduino.h>` in the top of your `main.cpp` file?

SimonRydell, 2016-06-16:
I removed it while trying to debug. Sorry, here comes the error produced with it

 [Thu Jun 16 23:18:29 2016] Processing nodemcuv2 (platform: espressif, board: nodemcuv2, framework: arduino)

-------------------------------------------------------------------------------
xtensa-lx106-elf-g++ -o .pioenvs/nodemcuv2/src/main.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-sect
ion-literals -falign-functions=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPL
ATFORMIO=021001 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDUINO=20200 -U__STRICT_ANSI__ -I/Users/Sim
onRydell/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I/Users/SimonRydell/.platformio/packages/fram
ework-arduinoespressif/tools/sdk/lwip/include -I.pioenvs/nodemcuv2/FrameworkArduino -I.pioenvs/nodemcuv2/FrameworkArduino
Variant -I.pioenvs/nodemcuv2/SoftwareSerial -I.pioenvs/nodemcuv2/ESP8266_ID127 -Isrc src/main.cpp
<command-line>:0:9: error: expected identifier before numeric constant
.pioenvs/nodemcuv2/ESP8266_ID127/ESP8266.h:38:7: note: in expansion of macro 'ESP8266'
class ESP8266 {
^
<command-line>:0:9: error: expected unqualified-id before numeric constant
.pioenvs/nodemcuv2/ESP8266_ID127/ESP8266.h:38:7: note: in expansion of macro 'ESP8266'
class ESP8266 {
^
<command-line>:0:9: error: expected unqualified-id before numeric constant
src/main.cpp:26:1: note: in expansion of macro 'ESP8266'
ESP8266 wifi(Serial1);
^
src/main.cpp: In function 'void setup()':
src/main.cpp:34:20: error: 'wifi' was not declared in this scope
Serial.println(wifi.getVersion().c_str());
^
xtensa-lx106-elf-gcc -o .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_wiring_pulse.o -c -std=gnu99 -Wpointer-arith -Wn
o-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-f
unctions=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=021001 -DESP8
266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDUINO=20200 -U__STRICT_ANSI__ -I/Users/SimonRydell/.platformio/p
ackages/framework-arduinoespressif/tools/sdk/include -I/Users/SimonRydell/.platformio/packages/framework-arduinoespressif
/tools/sdk/lwip/include -I.pioenvs/nodemcuv2/FrameworkArduino -I.pioenvs/nodemcuv2/FrameworkArduinoVariant .pioenvs/nodem
cuv2/FrameworkArduino/core_esp8266_wiring_pulse.c
xtensa-lx106-elf-gcc -o .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_wiring_pwm.o -c -std=gnu99 -Wpointer-arith -Wno-
implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-fun
ctions=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=021001 -DESP826
6 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDUINO=20200 -U__STRICT_ANSI__ -I/Users/SimonRydell/.platformio/pac
kages/framework-arduinoespressif/tools/sdk/include -I/Users/SimonRydell/.platformio/packages/framework-arduinoespressif/t
ools/sdk/lwip/include -I.pioenvs/nodemcuv2/FrameworkArduino -I.pioenvs/nodemcuv2/FrameworkArduinoVariant .pioenvs/nodemcu
v2/FrameworkArduino/core_esp8266_wiring_pwm.c
scons: *** [.pioenvs/nodemcuv2/src/main.o] Error 1
xtensa-lx106-elf-gcc -o .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_wiring_shift.o -c -std=gnu99 -Wpointer-arith -Wn
o-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-f
unctions=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=021001 -DESP8
266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDUINO=20200 -U__STRICT_ANSI__ -I/Users/SimonRydell/.platformio/p
ackages/framework-arduinoespressif/tools/sdk/include -I/Users/SimonRydell/.platformio/packages/framework-arduinoespressif
/tools/sdk/lwip/include -I.pioenvs/nodemcuv2/FrameworkArduino -I.pioenvs/nodemcuv2/FrameworkArduinoVariant .pioenvs/nodem
cuv2/FrameworkArduino/core_esp8266_wiring_shift.c
========================= [ ERROR ] Took 1.11 seconds =========================

SimonRydell, 2016-06-16:
The code I'm trying to upload:

`#include "Arduino.h"
#include "ESP8266.h"
#define SSID        "ITEAD"
#define PASSWORD    "12345678"

ESP8266 wifi(Serial1);

void setup(void)
{
    Serial.begin(9600);
    Serial.print("setup begin\r\n");

    Serial.print("FW Version: ");
    Serial.println(wifi.getVersion().c_str());

    if (wifi.setOprToStation()) {
        Serial.print("to station ok\r\n");
    } else {
        Serial.print("to station err\r\n");
    }

    if (wifi.joinAP(SSID, PASSWORD)) {
        Serial.print("Join AP success\r\n");
        Serial.print("IP: ");
        Serial.println(wifi.getLocalIP().c_str());
    } else {
        Serial.print("Join AP failure\r\n");
    }

    Serial.print("setup end\r\n");
}

void loop(void)
{
}`

ivankravets, 2016-06-16:
What do you want to do? You are going to use library that is written for Atmel AVR development platform where you use `espressif`.

This library http://platformio.org/lib/show/127/ESP8266 is intended to be used as Master (AVR) that will communicate with ESP8266 via UART.

ivankravets, 2016-06-16:
Please uninstall 127 library and take a look at these examples: https://github.com/esp8266/Arduino/tree/master/libraries

These are the native examples for your Node MCU

SimonRydell, 2016-06-16:
Oh, I see. I'm sorry, I'm new to this. So I can only use libraries which use the platform `espressif`. Thank you for the help!

ivankravets, 2016-06-16:
@SimonRydell be careful with "Compatibility" field near each library.

trevor-sonic, 2016-09-16:
"lib/private_lib/[here are source files]" structure for project related private library DOES NOT work I've tried every possibility . impossible. If I copy beside main.cpp it is working but in lib/MyLib  OR lib/src/MyLib.h   .cpp  unreachable !!
Atom seems beautiful but always errors ! version 1.9.9

ivankravets, 2016-09-17:
@trevor-sonic see examples in README.txt file that is located in `lib` folder

lpan, 2016-11-02:
@ivankravets I am trying include [Orbit_Oled_Library.zip](https://github.com/platformio/platformio/files/565640/Orbit_Oled_Library.zip) to my project which has a dir structure like this
```
├── ChrFont0.c
├── delay.c
├── delay.h
├── FillPat.c
├── FillPat.h
├── LaunchPad.h
├── OrbitBoosterPackDefs.h
├── OrbitOled.c
├── OrbitOledChar.c
├── OrbitOledChar.h
├── OrbitOledGrph.c
├── OrbitOledGrph.h
└── OrbitOled.h
```
How should I group the files so that I can use this library in my `main.cpp`? According to `/lib/readme.txt`, I grouped the .c and .h files with the same name. For example, I put `OrbitOledChar.c` and `OrbitOledChar.h` in a directory named `OrbitOledChar`. However, these files all depend on each other and platformio is not able to resolve them. For example, in `OrbitOled.c`, there are lines like
```c
#include "Launchpad.h"
#include "OrbitChar.h"
```
the compiler is not able to find these header files because they have been moved to a different directory. What should I do?

Thanks a lot

ivankravets, 2016-11-02:
You should put all files.

lpan, 2016-11-02:
@ivankravets thanks for your reply :smile: so I put all files in one single directory and I can name it whatever I want? I am very new to this field :stuck_out_tongue:

ivankravets, 2016-11-02:
Please check http://docs.platformio.org/en/stable/userguide/lib/cmd_install.html

You can install library directly from ZIP archive.

Hamza5, 2017-01-24:
Hello,
I am facing the same problem, no library is working for me, tried to install from internet (`pio lib install library_name`), tried to copy the `.h` and `.cpp` in `lib/library_name`, tried the global `~.platformio/lib` also but nothing worked ! 😞

ivankravets, 2017-01-24:
@Hamza5 which library do you install?

Hamza5, 2017-01-24:
@ivankravets 
For now I am not really interested to install a specific library but just to use [this file](https://gist.github.com/asheeshr/9004783), I transformed it into two files : `ardprint.c` and `ardprint.h` and put them into a folder named `ardprint` in the `lib` folder, it didn't work (`#include <ardprint.h>` gives an error). I thought that I didn't understand how to put them so I decided to install a random library, so I installed [SimpleTimer](http://platformio.org/lib/show/419/SimpleTimer/) and did `#include <SimpleTimer.h>` but the same problem happened. I tried also to use the global library folders but I got the same error. Why is not detecting the library files ?

I attached a screenshot of my current setup (returned to the first one) :

[image]

ivankravets, 2017-01-24:
@Hamza5 Menu > PlatformIO > Rebuild C/C++ Index?

Hamza5, 2017-01-24:
Wow finally it worked ! 😄  I tried to rebuild the index many times but it failed, but now It worked, and I figured out why : previously I opened a parent directory (`/home/hamza/PFE`) as my project directory, that is why it didn't work, but now I am opening `/home/hamza/PFE/Code/Arduino` and it works ! that was a very stupid problem.
Thanks a lot 👍

Disane87, 2018-01-02:
Problem seems to persists. I have this structure:
[image]

Locally it works fine with including the ``<API.h>`` but if I run the build within a Docker container,  the output looks like this:

``
Detected non-PlatformIO `lib_dir` option in `[env:lolin32]` section
Verbose mode can be enabled via `-v, --verbose` option
Collected 27 compatible libraries
Looking for dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <arduino-esp32>
|-- <WifiManager> v0.99.9
|   |-- <WebServer> v1.0
|   |   |-- <WiFi> v1.0
|   |   |-- <FS> v1.0
|   |-- <DNSServer> v1.1.0
|   |   |-- <WiFi> v1.0
|   |-- <WiFi> v1.0
|-- <WebServer> v1.0
|   |-- <WiFi> v1.0
|   |-- <FS> v1.0
|-- <DNSServer> v1.1.0
|   |-- <WiFi> v1.0
|-- <NTPClient> v3.1.0
|-- <ESP8266_SSD1306> v3.2.7
|   |-- <Wire> v1.0
|-- <Ticker> v2.0.2
|-- <ESP32httpUpdate> v2.1.145
|   |-- <WiFi> v1.0
|   |-- <HTTPClient> v1.1
|   |   |-- <WiFiClientSecure> v1.0
|   |   |   |-- <WiFi> v1.0
|   |   |-- <WiFi> v1.0
|   |-- <Update> v1.0
|   |-- <FS> v1.0
|   |-- <SPIFFS> v1.0
|   |   |-- <FS> v1.0
|-- <ArduinoJson> v5.12.0
|-- <Wire> v1.0
|-- <WiFi> v1.0
|-- <WiFiClientSecure> v1.0
|   |-- <WiFi> v1.0
|-- <ESPmDNS> v1.0
|-- <SPIFFS> v1.0
|   |-- <FS> v1.0
|-- <FS> v1.0
|-- <ArduinoOTA> v1.0
|   |-- <ESPmDNS> v1.0
|   |-- <Update> v1.0
|   |-- <WiFi> v1.0
Current build targets ['buildprog', 'size']
Board: lolin32
Compiling .pioenvs/lolin32/src/main.o
Archiving .pioenvs/lolin32/libFrameworkArduinoVariant.a
src/main.cpp:2:17: fatal error: API.h: No such file or directory

************************************************************
* Looking for API.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:API.h"
* Web  > http://platformio.org/lib/search?query=header:API.h
*
************************************************************

compilation terminated.
*** [.pioenvs/lolin32/src/main.o] Error 1
`` 

My platformio.ini:
```ini
[env:lolin32]
platform = espressif32
board = lolin32
framework = arduino

lib_dir = /lib/
extra_scripts = prepareBuild.py

lib_deps =
    https://github.com/espressif/arduino-esp32.git
    https://github.com/zhouhan0126/WIFIMANAGER-ESP32.git
    https://github.com/zhouhan0126/WebServer-esp32.git
    https://github.com/zhouhan0126/DNSServer---esp32.git
    https://github.com/taranais/NTPClient.git
    https://github.com/squix78/esp8266-oled-ssd1306.git
    1586
    2013
    64
```

Where is my error? Locally it works, with a fresh new docker container not.

ivankravets, 2018-01-02:
Unix has case sensitive file system

closed 2018-01-02 by ivankravets

Disane87, 2018-01-02:
Thank you! Totally forgot about this.

ivankravets, 2024-03-27:
Hi Abrishek (@Riz-waan), please forward your question to https://community.platformio.org/

commit e2795716 (2015-08-22): Generate "readme.txt" for project "lib" directory // Resolve #273